pal-robotics/aruco_ros #11: publishing of all detected markers

TN1ck, 2015-01-14:
There's currently the option to publish one ore two markers, but why not all of them?

bmagyar, 2015-01-14:
At the moment when I developed those nodes I needed to only have the selected ones published. 
In light of recent comments I am going to change this behaviour but add a filter that let's one configure if only a few are needed.

TN1ck, 2015-01-14:
Thanks for the fast reply!
Awesome! I can also fork the project and implement this behavior myself, I don't know how much time you have for this project. This project saved me in it's current state enough time that it would bee no problem for me :)

bmagyar, 2015-01-14:
If you wish to, go ahead with it. I will in this case only review and propose changes to the PR :)

phobon, 2015-01-14:
@TN1ck, I actually implemented this using the new native aruco message types in the `marker_pub` branch of my own fork, if you don't want to spend your own time implementing it.

TN1ck, 2015-01-15:
Oh wow, this is exactly what I wanted. Why don't you make a pull-request? Is there something missing in the branch?

bmagyar, 2015-01-15:
I urge you guys to do a PR, I'm gonna merge it and probably add it to my upcoming refactoring campaign of my old packages :)

phobon, 2015-01-15:
I will go ahead and do that then. Should be able to rebase it on master and submit the PR right away.

phobon, 2015-01-15:
PR #12 made.

bmagyar, 2015-01-20:
Since #12 is merged, I will close this issue.

closed 2015-01-20 by bmagyar