pal-robotics/aruco_ros #13: Marker Publisher Node Not Found

sauravag, 2015-03-31:
Hi,

Thank you for sharing this package! I tried to run the marker_publisher.launch file but I get the following error:

ERROR: cannot launch node of type [aruco_ros/marker_publisher]: can't locate node [marker_publisher] in package [aruco_ros]

I use the following command: $roslaunch marker_publisher.launch

I am able to get the single.launch file working perfectly and can see that the marker is detected. Also, the package was compiled without errors and installed in my catkin workspace.

Would you have any tips to fix this. 

Thanks!

jordi-pages, 2015-03-31:
@sauravag The marker_publisher executable was not include in the installation rule. Please pull your master branch and give it a try now.

sauravag, 2015-04-01:
Thanks Jordi!

I'll try it out :)

bmagyar, 2015-05-25:
I believe this is solved. Any feedback on it?

closed 2018-03-03 by bmagyar