pal-robotics/aruco_ros #15: Error - "process has died"

hecsaniz, 2015-05-18:
Hello,

Thanks for sharing this package. I am traying to work with Aruco and ROS and I have though that this package could be useful for me.

The problem is that I get the following error when I try to run the single.launch file:

laboratorio@ai2-labmix7:~/catkin_ws/src/aruco_ros/aruco_ros/launch$ roslaunch single.launch
... logging to /home/laboratorio/.ros/log/b49e99ec-fa18-11e4-895a-4c72b9b133b4/roslaunch-ai2-labmix7-13054.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ai2-labmix7:39150/

SUMMARY
========

PARAMETERS
 * /aruco_single/camera_frame
 * /aruco_single/image_is_rectified
 * /aruco_single/marker_frame
 * /aruco_single/marker_id
 * /aruco_single/marker_size
 * /aruco_single/reference_frame
 * /rosdistro
 * /rosversion

NODES
  /
    aruco_single (aruco_ros/single)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[aruco_single-1]: started with pid [13072]
[aruco_single-1] process has died [pid 13072, exit code -11, cmd /home/laboratorio/catkin_ws/devel/lib/aruco_ros/single /camera_info:=/stereo/left/camera_info /image:=/stereo/left/image_rect_color __name:=aruco_single __log:=/home/laboratorio/.ros/log/b49e99ec-fa18-11e4-895a-4c72b9b133b4/aruco_single-1.log].
log file: /home/laboratorio/.ros/log/b49e99ec-fa18-11e4-895a-4c72b9b133b4/aruco_single-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

I don't know how to solve this problem, so if anyone could help me it would be great.
Thank you.

bmagyar, 2015-05-18:
Hi,
There are not too many hints here that could help. Can you run the node in GDB so that we get a bit more insight?
Here is a wiki page that suggests some ways to do this from a launch file directly:
http://wiki.ros.org/roslaunch/Tutorials/Roslaunch%20Nodes%20in%20Valgrind%20or%20GDB

Start it with gdb, "run", then when it crashes execute "bt", and post the output here.

Also: what operating system, which ROS are you running on?

awesomebytes, 2015-05-18:
You could also start off executing the node with the output visible, just modify (or even better, make a new launch file based on the single.launch) the line:

https://github.com/pal-robotics/aruco_ros/blob/master/aruco_ros/launch/single.launch#L10

and add output="screen"

So it will look like:
    <node pkg="aruco_ros" type="single" name="aruco_single" output="screen">

So at least you'll get the prints of the node before dieing.

jordi-pages, 2015-05-18:
By default the node is subscribing to /stereo/left/camera_info and /stereo/left/image_rect_color. You should modify the single.launch file and remap the /image and /camera_info topics to the corresponding topics of your camera.

bmagyar, 2015-05-20:
I think I have ran into the issue today. Could you please confirm that the camera_info has D: [] ?

rostopic echo /your_camera/camera_info

hecsaniz, 2015-05-22:
Hi,

Sorry for no answering, I was very busy and I had to do a lot of things. I have tried many of the things that you told me to do (for example I had to remap the /image and /camera_info topics to my camera's topics), but I couldn't solve the problem.
Furthermore, I tried to use the package in another computer and it worked perfectly. So the solution was to uninstall ROS and then reinstall it.
Thank you for your quick answers, and thank you for the package, now it works very fine!

bmagyar, 2015-05-25:
I beleive I solved the problem that caused this crash. For now it is gone, the solution is not the best but lately I've been working on rearranging some code so it will get prettier.

https://github.com/pal-robotics/aruco_ros/commit/4c904f73493b913423df414f628698fc2d4435ea

closed 2015-05-25 by bmagyar

anirvan95, 2017-10-03:
Hi bmagyar. 

I am using aruco_ros and I am facing similar issue -  

[aruco_single-1] process has died [pid 23951, exit code -11, cmd /home/<user_name>/catkin_ws/devel/lib/aruco_ros/single /camera_info:=/usb_cam/camera_info /image:=/usb_cam/image_raw __name:=aruco_single __log:=/home/<user_name>/.ros/log/c47e5f6a-a839-11e7-8fb5-9c2a7029bf35/aruco_single-1.log].
log file: /home/<user_name>/.ros/log/c47e5f6a-a839-11e7-8fb5-9c2a7029bf35/aruco_single-1*.log 

I checked the log file, it was empty. I changed the topic to /usb_cam/camera_info. I checked the integrity of the camera_info and it was correct. I even reinstalled ROS indigo. 
I launched the node in debug mode wherein it says - 
Segmentation fault, in cv::Mat::setTo(cv::_InputArray const&, cv::_InputArray const&)()
from /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4

I am attaching the detailed error code in the picture. I will be thankful to you if you debug the issue as soon as possible.
[image]

Aya-nuomi, 2022-06-27:
> I am attaching the detailed error code in the picture. I will be thankful to you if you debug the issue as soon as possible.

Hello,
    I've met exactly the same problem as yours, do you work out it now?

> [aruco_single-2] process has died [pid 10981, exit code -6, cmd /home/z/my_ws/devel/lib/aruco_ros/single /camera_info:=/camera/color/camera_info /image:=/camera/color/image_raw __name:=aruco_single __log:=/home/z/.ros/log/5d2cf1b4-f5fa-11ec-bbc9-2811a83cf5b4/aruco_single-2.log].
log file: /home/z/.ros/log/5d2cf1b4-f5fa-11ec-bbc9-2811a83cf5b4/aruco_single-2*.log

    my log file is empty also, the only different is when I launched the node in debug mode, it says -

> Thread 1 "single" received signal SIGSEGV, Segmentation fault.
> 0x00007ffff67ffdc0 in cv::Size_<int>::Size_(cv::Size_<int> const&) ()
>    from /home/z/my_ws/devel/lib/libaruco.so

    here is the picture of my terminal:
[image]