pal-robotics/aruco_ros #19: Why do you have a rotation of axes when converting to TF in aruco_ros_utils?

balakumar-s, 2016-08-21:
It is in the arucMarker2Tf function.

bmagyar, 2016-08-22:
AFAIK I did that to publish a world-oriented coordinate frame, not a camera-oriented one. Camera frames often use Z as distance from camera pointing angle (depth) and X,Y align with the image axes. 

Having Z pointing up made life much simpler for applications using these markers such as object manipulation or robot localisation.

balakumar-s, 2016-08-22:
The ideal way to do that would be to set a TF transform between the camera frame and the world frame and then publishing the pose of the object with reference to the camera frame would get transformed. [TF setup link](http://wiki.ros.org/navigation/Tutorials/RobotSetup/TF ). I was setting up a common world frame between robot and camera and your rotations caused trouble and I removed the rotations and now it works well. Thanks for the ROS port.

closed 2016-08-22 by balakumar-s

bmagyar, 2016-08-23:
Yes you are right, although I weighted the most common usages and went for the internal rotation solution to save head scratching for most users.

kopp, 2017-02-26:
Actually, this started some head scratching for me: If I visualize the marker coordinate frame from the `result` topic with `image_view`, I get different coordinate systems than the rviz `tf` visualization:

[image]

(or is that a different issue?)

wollip, 2017-03-14:
the following code snipplet in c++ seems to fix the rotation issue for me. 

```
#define PI 3.14159265
#define TAU PI*2
#define REFANGLE(x) ((x>PI) ? x-TAU : (x<-PI) ? x + TAU : x ) 

            tf2::Quaternion q = to_ros_tf.getRotation();
            tf2::Matrix3x3 m(q);
            tf2::Vector3 translation( 0,0,0);
            double r, p, y;
            double a,b,c;
            m.getRPY(r,p,y);
            a = REFANGLE( p + PI );
            b = REFANGLE( -r - PI/2 );
            c = REFANGLE( y + PI/2 );
            m.setRPY(a,b,c);
            m.getRotation(q);
            to_ros_tf.setRotation(q);
```

where `to_ros_tf` is a `tf2::Transform` that is equivalent to the `aruco_msgs/Marker/pose/pose`

I plan on moving this fix directly to aruco ros. I will create a pull request when I have tested it formally

bmagyar, 2017-03-15:
One is the internal aruco visualisation, the tf frame is the output I wanted in 2012 :) 
There's no further explanation to that. 
If you want to fix that, feel free to send in a pull request.

kopp, 2017-04-03:
I'm fine with it -- I only use the ros tf output.  I primarily wanted to document this issue :-)