pal-robotics/aruco_ros #20: Markers not detected

mehditlili, 2016-08-30:
I switched from ar track alvar to aruco due to some issues with camera model and the matrix used to compute the marker pose but I am now having problems getting aruco to detect the markers. Here is one example of the thresholded image using default parameters:
[image]
Why is this marker not being detected??

bmagyar, 2016-08-30:
I think the one in the middle should work if the frame was a rectangle. On this image it looks more like a puzzle piece. Could you also attach the original image please?

As for the markers on the side, I'm not sure if this is not too much distortion...

Have you tried tweaking the parameters for your system using rqt_reconfigure?

jordi-pages, 2016-08-30:
Have you modified file single.launch and included the right marker ID and its proper size?

mehditlili, 2016-08-30:
Yes the marker id and size were set correctly, the markers on the bottom are irrelevant here (alvar markers). The frame shape around the marker should be irrelevant too?, it is just white paper fixed with white double sided tape on a metal piece.
[image]

mehditlili, 2016-08-30:
Here is two comparisons, there is a difference of 3 cm between both positions.
[image]
[image]

jordi-pages, 2016-08-30:
I would first try to detect another marker, just to make sure that AruCo does not have any problem with this specific marker.

bmagyar, 2016-08-31:
Good idea. Give a shot to the same scene with this marker (this is the most balanced one):
https://github.com/pal-robotics/aruco_ros/blob/master/aruco_ros/etc/marker582_5cm.jpg

As a sidenote: this coming weekend I intend to make a ROS Kinetic package of `aruco` with the new version of the library, that may solve your issue.

Flacedoo, 2017-02-02:
Is there a solution yet for the problem that markers are not recognized from a further than a specific distance? I ran a simulation, and when the camera is further away than 1m from a 6.5cm tag, it hasn't got any chance to detect it. Other libraries like apriltag has no problem at all with more than double the distance!