pal-robotics/aruco_ros #26: Kinetic support or install from source

matwilso, 2016-11-20:
Are there any plans to adapt this for kinetic?

I tried cloning and building from source, but it looks like there are problems with the opencv2 to opencv3 switch.  

I'm getting errors like:
```bash
.../ar_sys/src/aruco/cameraparameters.cpp:92:5: error: ‘Rodrigues’ is not a member of ‘cv’
     cv::Rodrigues(Rvec, m33)  ;
```
and 
```bash 
.../ar_sys/src/aruco/cvdrawingutils.cpp:139:9: error: ‘line’ is not a member of ‘cv’
         cv::line(Image,imagePoints[i+4],imagePoints[4+(i+1)%4],Scalar(0,0,255,255),1,CV_AA);
```

Thanks

bmagyar, 2016-11-20:
Yes I have a kinetic version planned.

bmagyar, 2017-09-22:
Why have you closed it?

matwilso, 2017-09-22:
I thought it was kind of a moot point, since it easy to build from source.  But if there are still plans, I will reopen it.

awesomebytes, 2018-04-20:
Any chance to have it released for kinetic?

awesomebytes, 2018-04-20:
For the record, I tried the `kinetic-devel` branch on a fresh Ubuntu 16.04 64b VM, did `catkin_make install` sourced that install folder, connected the image topic to a robot and it worked.

bmagyar, 2018-04-20:
It's on the way guys!

bmagyar, 2018-05-11:
Release was made, let us conclude this issue by closing it :)

closed 2018-05-11 by bmagyar

awesomebytes, 2018-05-11:
<3