pal-robotics/aruco_ros #38: aruco single not working on indigo

vannem95, 2017-12-13:
This was working fine on kinetic. I needed it on indigo. Here are the launch files and the errors i am getting. 
Anyone know what the issue might be?

Thanks in advance.

[image]
[image]

bmagyar, 2017-12-18:
Not really...Actually the current version should support Indigo more and is
less ideal for Kinetic.

LeLiPepsi, 2019-01-27:
Hi guys, I might have figured out the reason for this problem. It seems that we clone the pkg from kinetic branch instead of indigo branch. So the codes will not work in indigo. Just try the following command to clone the correct branch to workspace and compile again. It works for me. ;)

$ git clone -b indigo-devel https://github.com/pal-robotics/aruco_ros.git

closed 2020-09-17 by v-lopez