pal-robotics/aruco_ros #39: Ros wrapper for Aruco 3.0.x

xepost, 2017-12-13:
Hi ,

First of  all thanks for the  hard work. I would like to inquire  if you are planning to update the aruco library to  Aruco 3.0.X version  any time soon.

bmagyar, 2017-12-14:
Hiya, I had it on my table for a while now but not sure I will get to it
anytime soon. Are you willing to put up an initial PR perhaps?

xepost, 2017-12-15:
I have   another  4-5  packages  I need to sort out  first  but  I think I might  look after  new  year. I will keep you updated. Feel  free to  start earlier if you have time. The code is really faster  and  worth the  effort.

JaouadROS, 2018-02-06:
I created an ros package to detect markers using aruco 3, it is not really a wrapper. 
You've to install aruco lib in your machine then you can use the package. It is fast and detect all the libraries (ARUCO, ARTOOLKITPLUS, Apriltags.. and many others, check my launch file) with different size.

[here](https://github.com/JaouadROS/marker_publisher)

v-lopez, 2020-09-17:
melodic-devel branch now uses aruco 3.0.4

closed 2020-09-17 by v-lopez