pal-robotics/aruco_ros #40: Build fail when using catkin build instead of catkin_make

cthorey, 2017-12-21:
I got the library working nicely when build with **catkin_make**. 
However, when using **catkin build** instead **catkin_make** I got an error during the build

```
/usr/bin/ld: CMakeFiles/marker_publisher.dir/src/aruco_ros_utils.cpp.o: undefined reference to symbol '_ZN2cv9RodriguesERKNS_11_InputArrayERKNS_12_OutputArrayES5_'
/opt/ros/kinetic/lib/libopencv_calib3d3.so.3.3: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [/catkin_ws/devel/.private/aruco_ros/lib/aruco_ros/marker_publisher] Error 1
make[1]: *** [CMakeFiles/marker_publisher.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: CMakeFiles/double.dir/src/aruco_ros_utils.cpp.o: undefined reference to symbol '_ZN2cv9RodriguesERKNS_11_InputArrayERKNS_12_OutputArrayES5_'
/opt/ros/kinetic/lib/libopencv_calib3d3.so.3.3: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [/catkin_ws/devel/.private/aruco_ros/lib/aruco_ros/double] Error 1
make[1]: *** [CMakeFiles/double.dir/all] Error 2
/usr/bin/ld: CMakeFiles/single.dir/src/aruco_ros_utils.cpp.o: undefined reference to symbol '_ZN2cv9RodriguesERKNS_11_InputArrayERKNS_12_OutputArrayES5_'
/opt/ros/kinetic/lib/libopencv_calib3d3.so.3.3: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [/catkin_ws/devel/.private/aruco_ros/lib/aruco_ros/single] Error 1
make[1]: *** [CMakeFiles/single.dir/all] Error 2
make: *** [all] Error 2

```

This is happening during the build of **aruco_ros**, aruco and aruco_msg build is ok. 

It appears to be a problem with /libopencv_calib3d3.so.3.3 but I don t quiet know how to modify the CMakeLists.txt to fix it. 

Anyone had this problem before ?

Thx

bmagyar, 2017-12-21:
Is it possible that you have multiple opencv installations?

cthorey, 2017-12-21:
I don t see any when running 
```
 pkg-config opencv --libs
```

Voidminded, 2018-01-03:
You need to add `OpenCV` to `aruco_ros` package's dependencies and link all your executables to the library. 
I fixed it in [this fork](https://github.com/Voidminded/aruco_ros/tree/kinetic-devel):
`https://github.com/Voidminded/aruco_ros/tree/kinetic-devel`
Will send a PR as well.

cthorey, 2018-01-04:
thanks a lot, work like a charm.

bmagyar, 2018-01-04:
Closing as per #41

closed 2018-01-04 by bmagyar

nevermore0127, 2018-06-10:
/home/sia/catkin_handeye/devel/lib/aruco_ros/single: error while loading shared libraries: libopencv_calib3d3.so.3.1: cannot open shared object file: No such file or directory
[aruco_single-2] process has died [pid 9611, exit code 127, cmd /home/sia/catkin_handeye/devel/lib/aruco_ros/single /camera_info:=/stereo/right/camera_info /image:=/stereo/right/image_rect_color __name:=aruco_single __log:=/home/sia/.ros/log/73885db8-6c5b-11e8-82b0-2c946400b13d/aruco_single-2.log].
log file: /home/sia/.ros/log/73885db8-6c5b-11e8-82b0-2c946400b13d/aruco_single-2*.log
It seems I have the same error when I roslaunch aruco_ros/single in indigo-devel.what shoule i do for this? waiting for reply,thanks.

bmagyar, 2018-06-10:
@nevermore0127 please don't repost your problem everywhere. If you think you have the same error try applying the same solution.