pal-robotics/aruco_ros #45: Pose of the marker

ChunJyeBehBeh, 2018-02-21:
Hello ,

I am New to ROS and recently just build this package inside my workspace.

I follow the tutotial (http://sauravag.com/2015/03/how-to-tracking-aruco-marker-using-a-usb-webcam-and-ro/) but I didn't create the usb_cam-stream.launch (because I don't know put this file at where)

but I am unable to get the pose of the marker. Which topic should I echo? 

[image]

Tutorgaming, 2018-03-02:
1. the launch file can be placed anywhere and can be run using "roslaunch filename.launch" 
2. Can you show the result of "rostopic list" 

3. if you read the code you will find out that [the line 143](https://github.com/pal-robotics/aruco_ros/blob/541141679911a509d333896eb4594acbfda85129/aruco_ros/src/simple_single.cpp#L143)  indicates that it will publish the result in TF format 

you can try visualizing in Rviz to see the result 

PS. I haven't tried the aruco node yet, just guess the solution

bmagyar, 2018-03-03:
Please use [ROS answers](https://answers.ros.org/questions/) for questions.

closed 2018-03-03 by bmagyar