pal-robotics/aruco_ros #47: Does not build

villanuevab, 2018-03-27:
I keep getting the following error: 
```
ws/src/aruco_ros/aruco_ros/src/marker_publish.cpp:46:36: fatal error: aruco_msgs/MarkerArray.h: No such file or directory
compilation terminated.
aruco_ros/aruco_ros/CMakeFiles/marker_publisher.dir/build.make:62: recipe for target 'aruco_ros/aruco_ros/CMakeFiles/marker_publisher.dir/src/marker_publish.cpp.o' failed
make[2]: *** [aruco_ros/aruco_ros/CMakeFiles/marker_publisher.dir/src/marker_publish.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[100%] Linking CXX shared library /home/user/dev/catkin_ws/devel/lib/libaruco_ros_utils.so
CMakeFiles/Makefile2:16004: recipe for target 'aruco_ros/aruco_ros/CMakeFiles/marker_publisher.dir/all' failed
```
I think it may be a dependency issue, but I don't know how to get around it. Please advise. I am using ros kinetic.

awesomebytes, 2018-03-27:
Others have built it, I just gave it a try on Ubuntu 16.04 with a relatively fresh ROS Kinetic install:
```bash
mkdir -p aruco_ws/src
cd aruco_ws/src
git clone https://github.com/pal-robotics/aruco_ros
cd ..
rosdep install --from-path src --ignore-src
catkin_make
```

It worked for me.

bmagyar, 2018-03-27:
Try adding `add_dependencies` rules to all executables in `aruco_ros` against `aruco_ros_generate_messages_cpp`.

bmagyar, 2018-03-27:
... and please let us know if it works so we can update our cmake.

villanuevab, 2018-03-27:
I was able to resolve the issue. I have other packages that depend on aruco_ros and so wanted to build via: 
```
catkin_make --pkg aruco_ros
```
However, this was attempting to build `aruco_ros/aruco_ros`! 

I was able to resolve the issue by running `catkin_make` alone. Thank you for your quick responses.

closed 2018-03-27 by villanuevab