pal-robotics/aruco_ros #49: catkin_make have a error:can't find arucoConfig.cmake     aruco-config.cmake

liyi2017, 2018-04-17:
hi ,iam catkin_make the aruco_ros,but i have the error :+1: 
-- Could not find the required component 'aruco'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "aruco" with any of
  the following names:

    arucoConfig.cmake
    aruco-config.cmake

  Add the installation prefix of "aruco" to CMAKE_PREFIX_PATH or set
  "aruco_DIR" to a directory containing one of the above files.  If "aruco"
  provides a separate development package or SDK, be sure it has been
  installed.
i need some help,how to deal it? thx

bzdfzfer, 2018-05-11:
this problem can be fixed by "sudo apt install ros-kinetic-aruco"

bmagyar, 2018-05-11:
This issue was opened before the kinetic release. Please use the released version by installing from apt.

closed 2018-05-11 by bmagyar