pal-robotics/aruco_ros #51: Marker Plate wrong axis?

CesMak, 2018-05-29:
Hey there,

I just wonder the following:

In the picture it looks like the y axis (green) is going out of the red plate but actually this should be the blue z - axis or??

[image]

does that mean the marker axis are displayed wrong?!

ctaipuj, 2018-09-28:
Hi, @CesMak ,
We found out that the orientation issue is caused due to a rotation matrix described in **aruco_ros_utils.cpp** see [these code lines](https://github.com/pal-robotics/aruco_ros/blob/6e87d5d0e2da54f50fccb5c024d84f8150a33550/aruco_ros/src/aruco_ros_utils.cpp#L45-L67). You could modify the rotate_to_ros cv::Mat object and use it as an identity matrix to undo the rotation. (Actually not perform a rotation at all). So your TF data will be the same as the one found by ARuco.

If you change this take into account that the **simple_double.cpp** has to be modified too, in order to avoid the same issue (this only affects the simple_double.cpp file). So you will have to change [this line](https://github.com/pal-robotics/aruco_ros/blob/6e87d5d0e2da54f50fccb5c024d84f8150a33550/aruco_ros/src/simple_double.cpp#L99) to `mDetector.detect(inImage, markers, camParam, marker_size,false);` since the default value of `setYPerpendicular` is set to true in the .h file.

nevermore0127, 2018-10-23:
I have the same question about that. So witch axis is right ? Displayed in /aruco_single/result OR displayed in rviz ? 
 

> Hi, @CesMak ,
> We found out that the orientation issue is caused due to a rotation matrix described in **aruco_ros_utils.cpp** see [these code [...]

I have the same question about that and still don't understand by your reply. 
So witch axis are right ? Displayed in /aruco_single/result OR displayed in rviz ? 
Is the TF data right ? Or i should do things as you say then  the TF data will be right?
thank you

nevermore0127, 2018-10-23:
> modify the rotate_to_ros cv::Mat object and use it as an identity matrix to undo the rotation

When I modify the rotate_to_ros cv::Mat object and use it as an identity matrix to undo the rotation , the axis seems not the same . And i modify the matrix to 
-1 0 0
 0 -1 0
 0 0 1
the axis looks same as aruco_single/result displayed , but why? 
[image]

v-lopez, 2018-10-23:
You can also test with the ARuco 3.0.4 branch: https://github.com/pal-robotics/aruco_ros/tree/aruco-3.0.4

I tried to adapt the package conventions to the library convention. The problem is that if we merged this, a lot of code that relies on the orientation would stop working as expected.

ctaipuj, 2018-10-31:
@nevermore0127 
Answering your questions, 
>which axis are right? 

The Z axis (the blue one) must be "going out" of the ARuco tag.
> Displayed in /aruco_single/result OR displayed in rviz ?

So following the ARuco convention the /aruco_single/result has the correct orientation. Notice that this not affect translation at all.

>Is the TF data right ?

As I said above, the orientation issue doesn't affect the pose estimation. So, yes the TF data is right the only difference is that it has and additional rotation. So in Rviz you see the y-axis pointing out the tag.

>Or i should do things as you say then the TF data will be right?

Is not necessary it is just for convention, according to your needs you may need a different orientation. Or you can publish a static transform with a new rotation. (this is is the simpliest solution).

Notice that the TF data is only correct if your camera is correctly calibrated.

If I am not clear enough, please let me know.

nevermore0127, 2018-11-05:
> @nevermore0127
> Answering your questions,
> 
> > which axis are right?
> 
> The Z axis (the blue one) must be "going out" of the ARuco tag.
> 
> > Displayed in /aruco_single/result OR displayed in rviz ?
> 
> So following the ARuco convention the /aruco_single/result has the correct orientation. Notice that this not affect translation at all.
> 
> > Is the TF data right ?
> 
> As I said above, the orientation issue doesn't affect the pose estimation. So, yes the TF data is right the only difference is that it has and additional rotation. So in Rviz you see the y-axis pointing out of the tag.
> 
> > Or i should do things as you say then the TF data will be right?
> 
> Is not necessary it is just for convention, according to your needs you may need a different orientation. Or you can publish a static transform with a new rotation. (this is is the simpliest solution).
> 
> Notice that the TF data is only correct if your camera is correctly calibrated.
> 
> If I am not clear enough, please let me know.

Thanks,clear about that.

7675t, 2019-04-12:
Hi, I also found it is a bit confusing the TF axis direction is different from the detection result. I made a PR for this problem.

hhn1n15, 2019-04-21:
> You can also test with the ARuco 3.0.4 branch: https://github.com/pal-robotics/aruco_ros/tree/aruco-3.0.4
> 
> I tried to adapt the package conventions to the library convention. The problem is that if we merged this, a lot of code that relies on the orientation would stop working as expected.

I tried that and edited as you said but there is another error: 

Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "camera_rgb_optical_frame" from authority "unknown_publisher" because of an invalid quaternion in the transform (-0.483269 0.495631 0.381370 0.526505)
         at line 257 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp

Rviz then did not display camera frames at all. How to solve it?

zhudelong, 2019-04-24:
> @nevermore0127
> Answering your questions,
>
>
> The Z axis (the blue one) must be "going out" of the ARuco tag.
>
>
> So following the ARuco convention the /aruco_single/result has the correct orientation. [...]

@ctaipuj 
To make it more clear, if we want to use tf to transform a vector from marker frame to camera frame, we  have to use the frame in Rviz. For example, I specify the four corners in the marker frame, [[-0.5, 0, -0.5], [-0.5, 0, 0.5], [0.5, 0, -0.5], [0.5, 0, 0.5]], then use the tf to transform it to camera frame and get a correct result. if I use the result frame, it is wrong. 

by the way, how can I directly get the corner position?