pal-robotics/aruco_ros #52: issue with catkin_make

aashi7, 2018-05-31:
Hi,

For ROS-Kinetic, it seems that most tutorials mention a very simple procedure by copying to /catkin_ws/src and running catkin_make

as shown here
https://github.com/eYSIP-2017/eYSIP-2017_Navigation-in-Indoor-Environments-using-drone/wiki/Setup-ArUco-Mapping-package
and
https://github.com/warp1337/ros_aruco

Unfortunately, after long hours of trying, this still leads to "aruco_msgs" missing configuration file, or 'aruco'

Following
https://github.com/pal-robotics/aruco_ros/issues/49
and installing aruco, aruco-ros, aruco-msgs, and aruco-mapping from apt, i.e. sudo apt install ros-kinetic-aruco

but once these are installed from apt, compiling another library leads to issues.
We are trying to compile 
https://github.com/ankitdhall/lidar_camera_calibration
and it leads to 'lidar_camera_calibration" missing configuration file when running catkin_make
yet it works fine without that folder in the src folder. 

Anyone have a similar experience? 

We tried building non-aruco libraries from src with catkin_make, and it was successful. Our feeling is that something goes wrong with aruco builds, hence it can't even make them from source but can only do so after installing from apt.

Thanks,

bmagyar, 2018-05-31:
Please post the exact error.

NalinLuthra, 2018-11-07:
> it leads to 'lidar_camera_calibration" missing configuration file when running catkin_make
yet it works fine without that folder in the src folder.

It must a warning instead of error and it can still work fine without configuration file

NalinLuthra, 2018-11-07:
If you can attach a screenshot it would be helpful

closed 2018-11-08 by aashi7