pal-robotics/aruco_ros #55: aruco_ros/single: error while loading shared libraries: libopencv_calib3d3.so.3.1

nevermore0127, 2018-06-10:
catkin_handeye/devel/lib/aruco_ros/single: error while loading shared libraries: libopencv_calib3d3.so.3.1: cannot open shared object file: No such file or directory
[aruco_tracker-2] process has died [pid 12577, exit code 127, cmd /home/sia/catkin_handeye/devel/lib/aruco_ros/single /camera_info:=/kinect2/hd/camera_info /image:=/kinect2/hd/image_color_rect __name:=aruco_tracker __log:=/home/sia/.ros/log/1b49175c-6c63-11e8-ae99-2c946400b13d/aruco_tracker-2.log].
log file: /home/sia/.ros/log/1b49175c-6c63-11e8-ae99-2c946400b13d/aruco_tracker-2*.log
I try this on indigo-devel ,how can I solve this?

bmagyar, 2018-06-10:
Here: http://bfy.tw/IXG1

closed 2018-06-10 by bmagyar

nevermore0127, 2018-06-11:
Sorry，I changed the makefile add version 2.4.8 to find_package opencv ,and it seems fine now.