pal-robotics/aruco_ros #57: Marker Dictionary

Asif6511, 2018-09-27:
Hi,

Is it possible to use a different Aruco dictionary in this package? I know that now the package can detect markers from Original aruco dictionary. Please suggest a way  to implement the detection on a different Aruco dictionary. Thanks.

NalinLuthra, 2018-11-07:
Yes, You can easily use different dictionary just need to do some tweaks in the code it self. It will be easy if you would like to tell which dictionary ?

alanprodam, 2019-01-17:
Hi,

I have the same doubt, I would like to use the Dictionary: 4x4(50).

fhln, 2019-05-20:
yah...the same question

Sinchiguano, 2019-08-11:
same question but nor answer at all, anyone who solved the problem, let me know please.

DharminB, 2020-02-01:
I think the dictionary used is 5x5. [This function](https://github.com/pal-robotics/aruco_ros/blob/kinetic-devel/aruco/src/aruco/arucofidmarkers.cpp#L316) is pretty much hard coding it. I haven't tried changing it but that might be a good place to start. :man_shrugging:

xhqnogit, 2023-10-23:
Hello everyone, has anyone solved this problem? I wanted to use the DICT_7X7_100 dictionary, but I couldn't find it. Thank you so much for helping me

ErikParkerrr, 2024-01-08:
any updates on this?