pal-robotics/aruco_ros #61: tf::TransformBroadcaster didn't broadcast the transform of the marker to /tf

ancorasir, 2019-01-16:
I'm using ros kinetic and realsense. The package successfully locate the marker as I checked /aruco_single/result and /aruco_single/transform. But the strange thing is /tf topic is empty although "br.sendTransform(stampedTransform)" in the code. Has any one met this problem before? Thanks a lot.

g1y5x3, 2019-07-29:
I had a similar issue. Have you find out why this would happen?

TaekedeHaan, 2020-05-14:
I experience the same. It is fixed when I echo the marker or an other topic on the terminal (in my case `rostopic echo /aruco_tracker/pose`). The issue seems to be that the transform is only published when a subscriber is detected, as can be seen  [here](https://github.com/pal-robotics/aruco_ros/blob/959fe5fcdc0148e358a180e0366b79bde08c74a8/aruco_ros/src/simple_single.cpp#L186). Thus I think the only solution is to subscribe to one of these topics.

QuantuMope, 2021-04-11:
Is there anyway to have this work without manually subscribing to the pose?

saikishor, 2021-04-12:
@QuantuMope as @TaekedeHaan pointed out. The transform is only published when there is a subscription to one of the topics. It was designed such a way, as the ARuCo marker detection is computationally intensive(#92). 

You could write a ROS node that subscribes to one of the topics, at the instance you need the transform published.

dHonerkamp, 2022-04-11:
It would be great to mention this in the readme, as it seems that every new person (including myself) first spends a few hours figuring this out

wangqingyu985, 2022-10-24:
i also encountered the same issue