pal-robotics/aruco_ros #63: Cannort find Kinect2_rgb_optical frame

chandanlal92, 2019-04-03:


chandanlal92, 2019-04-03:
[ERROR] [1554290012.244204703]: Unable to get pose from TF: canTransform: source_frame kinect2_rgb_optical_frame does not exist.. canTransform returned after 0.502939 timeout was 0.5.
[ERROR] [1554290012.757058040]: Unable to get pose from TF: canTransform: source_frame kinect2_rgb_optical_frame does not exist.. canTransform returned after 0.504677 timeout was 0.5.
[ERROR] [1554290013.268903816]: Unable to get pose from TF: canTransform: source_frame kinect2_rgb_optical_frame does not exist.. canTransform returned after 0.503324 timeout was 0.5.

chandanlal92, 2019-04-03:
<launch>

    <arg name="markerId"        default="582"/>
    <arg name="markerSize"      default="0.034"/>    
    <arg name="eye"             default="left"/>
    <arg name="marker_frame"    default="aruco_marker_frame"/>
    <arg name="ref_frame"       default=""/>  
    <arg name="corner_refinement" default="LINES" /> 
     
    <include file="$(find kinect2_bridge)/launch/kinect2_bridge.launch" >
        <arg name="depth_device" value="0" />
    </include>

    <node pkg="aruco_ros" type="single" name="aruco_single">
        <remap from="/camera_info" to="/kinect2/qhd/camera_info" />
        <remap from="/image" to="/kinect2/qhd/image_color" />
        <param name="image_is_rectified" value="True"/>
        <param name="marker_size"        value="0.143"/>
        <param name="marker_id"          value="26"/>
        <param name="reference_frame"    value="kinect2_link"/>   
        <param name="camera_frame"       value="kinect2_rgb_optical_frame"/>
        <param name="marker_frame"       value="camera_marker" />
        <param name="corner_refinement"  value="$(arg corner_refinement)" />
    </node>

</launch>

chandanlal92, 2019-04-03:
This is my launch file

chandanlal92, 2019-04-03:
The Marker is detected properly. I am not able to see my kinect2_optical_frame

chandanlal92, 2019-04-03:
This will affect my calibration results

1gen100, 2019-06-24:
Did you find a solution to this problem? I also getting the same problem:

[ERROR] [1561377304.973332415]: Unable to get pose from TF: canTransform: source_frame kinect2_rgb_optical_frame does not exist.. canTransform returned after 0.502502 timeout was 0.5.
[ERROR] [1561377305.506447283]: Unable to get pose from TF: canTransform: source_frame kinect2_rgb_optical_frame does not exist.. canTransform returned after 0.503137 timeout was 0.5.
[ERROR] [1561377306.033191667]: Unable to get pose from TF: canTransform: source_frame kinect2_rgb_optical_frame does not exist.. canTransform returned after 0.500643 timeout was 0.5.

chandanlal92, 2019-06-24:
<param name="reference_frame"    value="kinect2_rgb_optical_frame"/>   
    <param name="camera_frame"       value="kinect2_rgb_optical_frame"/>

This should be the parameter...And Open rqt_image_view and choosr topic aruco_ros/result....If you do this you will get the frame.

1gen100, 2019-06-24:
Thank you very much, it work I'm not getting any ERRORs anymore :+1:

closed 2019-06-24 by chandanlal92