pal-robotics/aruco_ros #70: Compilation Error while running catkin_make

rahulmnitk, 2019-05-20:
rahul@Rahul:~/aruco_drone_ws$ catkin_make
Base path: /home/rahul/aruco_drone_ws
Source space: /home/rahul/aruco_drone_ws/src
Build space: /home/rahul/aruco_drone_ws/build
Devel space: /home/rahul/aruco_drone_ws/devel
Install space: /home/rahul/aruco_drone_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/rahul/aruco_drone_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/rahul/aruco_drone_ws/build"
####
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_lisp
[  2%] Built target aruco
[  2%] Built target geometry_msgs_generate_messages_nodejs
[  2%] Built target std_msgs_generate_messages_nodejs
[  2%] Built target _aruco_msgs_generate_messages_check_deps_Marker
[  2%] Built target _aruco_msgs_generate_messages_check_deps_MarkerArray
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_RC
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_HeadingCommand
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_ThrustCommand
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_ServoCommand
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_MotorCommand
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_AttitudeCommand
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_Compass
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_YawrateCommand
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_HeightCommand
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_Supply
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_VelocityZCommand
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_RawImu
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_VelocityXYCommand
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_RuddersCommand
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_RawRC
[  2%] Built target _cvg_sim_msgs_generate_messages_check_deps_MotorStatus
[  3%] Built target pal_vision_segmentation_gencfg
[  3%] Building CXX object pal_vision_segmentation/CMakeFiles/pal_vision_segmentation.dir/src/image_processing.cpp.o
[  3%] Built target _cvg_sim_msgs_generate_messages_check_deps_Altitude
[  3%] Built target _cvg_sim_msgs_generate_messages_check_deps_MotorPWM
[  3%] Built target _cvg_sim_msgs_generate_messages_check_deps_ControllerState
[  3%] Built target _cvg_sim_msgs_generate_messages_check_deps_RawMagnetic
[  3%] Performing update step for 'ardronelib'
[  3%] Built target _cvg_sim_msgs_generate_messages_check_deps_Altimeter
[  3%] Built target _cvg_sim_msgs_generate_messages_check_deps_PositionXYCommand
[  3%] Performing configure step for 'ardronelib'
No configure
[  3%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_time
[  3%] Performing build step for 'ardronelib'
[  3%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_adc_data_frame
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
[  3%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_pressure_raw
[  3%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_altitude
[  3%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_watchdog
[  3%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_euler_angles
Libs already extracted
Building target static
[  3%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_references
[  3%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_demo
[  3%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_video_stream
[  3%] Built target _ardrone_autonomy_generate_messages_check_deps_RecordEnable
[  3%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_zimmu_3000
[  3%] Built target _ardrone_autonomy_generate_messages_check_deps_navdata_trims
Architecture x86_64 is already built
Creating universal static lib file from architectures x86_64
[... 517 lines cut ...]
[  4%] Built target tf_generate_messages_cpp
[  4%] Built target tf_generate_messages_lisp
[  4%] Built target tf2_msgs_generate_messages_nodejs
[  4%] Built target actionlib_msgs_generate_messages_cpp
pal_vision_segmentation/CMakeFiles/pal_vision_segmentation.dir/build.make:62: recipe for target 'pal_vision_segmentation/CMakeFiles/pal_vision_segmentation.dir/src/image_processing.cpp.o' failed
make[2]: *** [pal_vision_segmentation/CMakeFiles/pal_vision_segmentation.dir/src/image_processing.cpp.o] Error 1
CMakeFiles/Makefile2:2866: recipe for target 'pal_vision_segmentation/CMakeFiles/pal_vision_segmentation.dir/all' failed
make[1]: *** [pal_vision_segmentation/CMakeFiles/pal_vision_segmentation.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  4%] Built target actionlib_generate_messages_nodejs
[  4%] Built target actionlib_generate_messages_lisp
[  4%] Built target tf_generate_messages_py
[  4%] Built target actionlib_msgs_generate_messages_eus
[  4%] Built target tf_generate_messages_nodejs
[  4%] Built target actionlib_msgs_generate_messages_lisp
Building ARDroneTool/Lib
[  5%] Performing install step for 'ardronelib'
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
[  5%] Completed 'ardronelib'
[  6%] Built target ardronelib
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

I have looked at the available solutions for this issue but still not able to figure out the problem. I have ran the rosdep command as well to check for dependencies and it says all dependencies have been installed correctly. I am using Kinect version of ROS. Please help. Thanks.

v-lopez, 2019-05-20:
I believe the compilation problem you are having is in https://github.com/pal-robotics/pal_vision_segmentation

I don't think that repository was ever migrated to ROS Kinetic, and probably is targeting an old OpenCV version.

v-lopez, 2019-05-20:
You don't need pal_vision_segmentation for aruco_ros.

If you want to use pal_vision_segmentation in Kinetic, you could migrate it to OpenCV3, we'd gladly accept a PR with those changes, here's some related documentation:
http://wiki.ros.org/kinetic/Migration#OpenCV

rahulmnitk, 2019-05-20:
I didn't get exactly what you meant by migrating it, but can I use aruco_ros for marker detection in a drone on Kinetic? I am following this tutorial which is meant for ROS Indigo.
https://github.com/eYSIP-2017/eYSIP-2017_Navigation-in-Indoor-Environments-using-drone/wiki/Installation
https://github.com/eYSIP-2017/eYSIP-2017_Navigation-in-Indoor-Environments-using-drone/wiki/ArUco-marker-pose-detection

v-lopez, 2019-05-20:
You should change the code so that the calls to OpenCV2 are replaced with calls to OpenCV3.

rahulmnitk, 2019-05-20:
Oh okay got it! Thanks a lot for your help.

closed 2019-05-20 by rahulmnitk