pal-robotics/aruco_ros #72: The marker generator

TFLQW, 2019-06-30:
Thanks for your contribution! But i still have some questions
How did you generate the ar marker and arrange the id to each marker, did you use the ros package ar_marker_alvar?

bmagyar, 2019-06-30:
`rosrun aruco optimalmarkers`

TFLQW, 2019-07-04:
@bmagyar how could input the parameter of the marker I want to generate like 
rosrun aruco optimalmarkers ...(parameters) what is the format of this command?

XYudong, 2019-10-10:
same question. Could you provide an example of the parameters

XYudong, 2019-10-10:
Hope to get an example.
BTW, `rosrun aruco optimalmarkers 4 0 340 xxx 22`
The "340" means the size of the output marker. others I don't know yet.

Wang-png, 2019-10-25:
http://chev.me/arucogen/

XYudong, 2019-10-25:
> http://chev.me/arucogen/

Hi,
But what generated by this seems different from `rosrun aruco optimalmarkers 4 0 340 xxx 22`, which may result in inconsistency issue in detection?

closed 2019-11-08 by TFLQW