pal-robotics/aruco_ros #74: Position of marker is significantly inaccurate across different cameras, despite good calibration

sngweicong, 2019-08-09:
OS: Ubuntu 16.04,
Kernel: 4.15.0
Cameras: Kinect V2, Realsense D435, Realsense D415

[image]

[image]

Markers are detecting fine, pose within 2D seems to be captured fine. Marker size (0.09 meter) is definitely correctly specified in the launch file. 3D pose is however, off by some translation, always further than the ground truth. Magnitude of the bias increases on the distance in a non-linear manner, meaning that if the marker is near, I probably have to multiply 0.98 to the returned position of the marker in the rgb camera frame to get close to the ground truth, and when it is far, I will have to multiply 0.94. I have verified that the point clouds are accurate by measuring the distances with a measuring tape. This occurs to all of my 1 x Kinect V2, 1 x Realsense D435 and 4 x Realsense D415.

When I showed my colleagues this problem, their immediate guesses were that the calibrations are off. After trying recalibration from multiple libraries, where I use
https://github.com/code-iai/iai_kinect2/tree/master/kinect2_calibration for the Kinect V2,
http://wiki.ros.org/visp_camera_calibration?distro=kinetic and
https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API for the Intel Realsense D435 and D415,
The problem remains, while point cloud registration looks perfect for all of my 6 cameras. Correct if I am wrong but if there is no issue of any kind with the point clouds, is it necessary true that both the intrinsics and the extrinsics are good?

I am stuck at this problem for at least a month and at my wits end, and I am using ar_track_alvar for now which is inaccurate due to how it gets the marker pose from point cloud instead of 2D image feed.

I would really appreciate advice of any kind, I am not familiar with camera equations, so I'd wager there is a good chance I am missing out on something extremely obvious here.

v-lopez, 2020-04-30:
I apologize for the huge delay. 

I hope you found your answer, If it happens among all your cameras I would suggest reviewing that the marker size is exactly what you are passing as parameter, and if not check the https://github.com/pal-robotics/aruco_ros/tree/aruco-3.0.4 branch which has a newer version of the aruco library.

closed 2020-04-30 by v-lopez

DavidYaonanZhu, 2020-07-23:
I found the same issue. Is there any solution?

DavidYaonanZhu, 2020-07-23:
@sngweicong