pal-robotics/aruco_ros #75: Is there a maximum detection distance?

dj4park, 2019-08-16:
I noticed that the aruco markers are no longer detected when the camera is any further than ~1.5m. When the camera is within the range of 1.5~1.8m, the entire marker and the surrounding white border is still visible. Is there a configuration parameter that I can change? What exactly is causing this constraint?

I am using a ZED camera which is calibrated.

Greg9914, 2020-03-27:
Hi, I have the very same question. Did you have any advance on improving it?

v-lopez, 2020-04-30:
I believe the only limitation is the resolution of the image. I've experienced using markers with less pixels and have found an increase in the detection range.

But we do not add an artifical limitation to the distance, I've measured markers over 4m with a raspicam and a 0.12x0.12m aruco marker.

closed 2020-04-30 by v-lopez