pal-robotics/aruco_ros #77: Aruco does not detect any marker.

Juliocasta5, 2020-03-26:
Hi,
i'm testing the aruco_ros package in Tiago robot and i'm trying to run the aruco_ros simulation following the wiki ros tutorials of Tiago. So, first, I launch the Tiago's simulation:

roslaunch tiago_gazebo tiago_gazebo.launch public_sim:=true robot:=steel world:=tutorial_office gzpose:="-x -3.5 -y -0.85 -z 0 -R 0 -P 0 -Y -3.0"

[image]
Then, you can see the robot in front of the aruco board, the next step is to launch the detector.

**rosrun aruco_ros single /camera_info:=/xtion/rgb/camera_info /image:=/xtion/rgb/image_rect_color _marker_id:=250  _marker_size:=0.6 _marker_frame:=/aruco_frame _camera_frame:=/xtion_optical_frame _image_is_rectified:=true _reference_frame:=/base_footprint**

After launching that command, it appears this in the terminal.
**[ INFO] [1585238075.740888622, 50.626000000]: Corner refinement method: 3
[ INFO] [1585238075.740995594, 50.626000000]: Threshold method: 1
[ INFO] [1585238075.741038275, 50.626000000]: Threshold method:  th1: 7 th2: 7
[ INFO] [1585238075.741066778, 50.626000000]: Marker size min: 0.04  max: 0.5
[ INFO] [1585238075.741087506, 50.626000000]: Desired speed: 0
[ INFO] [1585238076.041515145, 50.680000000]: Aruco node started with marker size of 0.600000 m and marker id to track: 250
[ INFO] [1585238076.041839764, 50.680000000]: Aruco node will publish pose to TF with /base_footprint as parent and /aruco_frame as child.
[ WARN] [1585238076.041961474, 50.680000000]: normalizeImageIllumination is unimplemented!**

I don't understand what's happening, if someone can help me please.

Horizonzhou, 2020-04-30:
I have the same issue, could you please share your experience about this issue?

v-lopez, 2020-04-30:
I believe you have changed the parameters for the aruco_ros single node.

The correct parameters are:
`rosrun aruco_ros single /camera_info:=/xtion/rgb/camera_info /image:=/xtion/rgb/image_rect_color _marker_id:=584  _marker_size:=0.10 _marker_frame:=/aruco_frame _camera_frame:=/xtion_optical_frame _image_is_rectified:=true _reference_frame:=/base_footprint`

You'll notice a different marker_id and marker_size.

With these parameters, you'll see that a tf frame appears on the location of the detected marker, as well as an rviz marker in the `/aruco_simple/marker` topic.

closed 2020-04-30 by v-lopez