pal-robotics/aruco_ros #79: ARuco tf position does not match actual

Circle-Kai, 2020-05-11:
I placed ARuco in front of the camera, but when RViz looked at the tf, the marker_frame was above the camera.
```
<launch>
  <arg name="markerId"          default="101"/>
  <arg name="markerSize"        default="0.045"/> <!-- in m -->
  <arg name="marker_frame"      default="aruco_marker_$(arg markerId)"/>
    
  <arg name="camera"            default="/camera/color"/>
  <arg name="camera_frame"      default="camera_link"/>
    
  <arg name="ref_frame"         default="/base_link"/> <!-- leave empty and the pose will be published wrt param parent_name -->
  <arg name="corner_refinement" default="LINES"/> <!-- NONE, HARRIS, LINES, SUBPIX -->

  <node pkg="aruco_ros" type="single" name="aruco_single">
    <remap from="/camera_info" to="$(arg camera)/camera_info" />
    <remap from="/image" to="$(arg camera)/image_raw" />
    <param name="camera_frame"       value="$(arg camera_frame)"/>
    
    <param name="image_is_rectified" value="True"/>
    <param name="marker_id"          value="$(arg markerId)"/>
    <param name="marker_size"        value="$(arg markerSize)"/>
    <param name="reference_frame"    value="$(arg ref_frame)"/> <!-- frame in which the marker pose will be refered -->  
    <param name="marker_frame"       value="$(arg marker_frame)" />
    <param name="corner_refinement"  value="$(arg corner_refinement)" />
  </node>
</launch>
```

sakthivelj, 2020-06-25:
> I placed ARuco in front of the camera, but when RViz looked at the tf, the marker_frame was above the camera.
> 
> ```
> <launch>
>   <arg name="markerId"          default="101"/>
>   <arg name="markerSize"        default="0.045"/> 
>   <arg name="marker_frame"      default="aruco_marker_$(arg markerId)"/>
>     
>   <arg name="camera"            default="/camera/color"/>
>   <arg name="camera_frame"      default="camera_link"/>
>     
>   <arg name="ref_frame"         default="/base_link"/> 
>   <arg name="corner_refinement" default="LINES"/> 
> 
>   <node pkg="aruco_ros" type="single" name="aruco_single">
>     <remap from="/camera_info" to="$(arg camera)/camera_info" />
>     <remap from="/image" to="$(arg camera)/image_raw" />
>     <param name="camera_frame"       value="$(arg camera_frame)"/>
>     
>     <param name="image_is_rectified" value="True"/>
>     <param name="marker_id"          value="$(arg markerId)"/>
>     <param name="marker_size"        value="$(arg markerSize)"/>
>     <param name="reference_frame"    value="$(arg ref_frame)"/>   
>     <param name="marker_frame"       value="$(arg marker_frame)" />
>     <param name="corner_refinement"  value="$(arg corner_refinement)" />
>   </node>
> </launch>
> ```

I

Eswar1991, 2020-06-25:
We are also facing this issue. Please see the image below:
[image]
As shown above, the location of marker frame in RViz does not match with that of gazebo. This really confuses us because we are not clear which dictates the physical location of marker in reality

Any help on this issue would be appreciated.

Eswar1991, 2020-06-25:
> Either the marker size is wrong or the camera is not properly calibrated.

But the calibration info is obtained from the standard topic of SUMMIT XL robot "camera info". We are not clear on how the marker size influence frame location, because the marker frame is located at a distance from camera frame in gazebo, whereas in rviz it is showing above. 

The correlation between marker frame location and marker size is not clear

v-lopez, 2020-06-25:
The size parameter for aruco determines the distance from the camera.

A 100px x 100px marker in the camera, could be a 10cm marker a 10 meters, or a 1cm marker at 1meter.

lalazari, 2021-09-07:
> The size parameter for aruco determines the distance from the camera.
> 
> A 100px x 100px marker in the camera, could be a 10cm marker a 10 meters, or a 1cm marker at 1meter.

I changed my aruco size in launch file from 0.12 to 0.13 and everything is fine. Thanks a lot!

saikishor, 2022-04-01:
Hello,

We will proceed to close this issue due to inactivity,

Thank you,

Best Regards,
PAL Robotics Team.

closed 2022-04-01 by saikishor