pal-robotics/aruco_ros #83: Robot localization using aruco marker in gazebo

Eswar1991, 2020-06-25:
We are trying to localize the robot in gazebo using aruco marker. We have successfully loaded the marker into gazebo, detected it, and obtained the transformation between camera frame and marker frame using aruco_ros (https://github.com/pal-robotics/aruco_ros) package. The transform between them obtained from tf package is in excellent agreement with the one obtained from the package.
[image]
However, the physical location of marker frame is not the same when we visualize it in RviZ. Please see the image below
[image]
As shown above, the location of marker frame in RViz does not match with that of gazebo. This really confuses us because we are not clear which dictates the physical location of marker in reality

Any help on this issue would be appreciated.
Thanks

chebbal, 2022-05-18:
@Eswar1991 did you find a fix for this issue??

saikishor, 2022-05-19:
Hello @Eswar1991 @chebbal,

 what version are you using for this?, We have recently fixed a bug in the implementation pointed in this issue : https://github.com/pal-robotics/aruco_ros/issues/98. May be this might be the root cause in your case too. Please let us know if this fix fixes your issue.
 
 Thank you,
 
 Best Regards,
 PAL Robotics Team.

chebbal, 2022-05-19:
Hi @saikishor ,

Thanks for the quick revert
In my case the issue was with frame definition, resolved it, it works fine now

regards,
Chaitanya

saikishor, 2022-05-19:
Hello @chebbal ,

Thank you for confirming that, then I will be proceeding to close this issue. 

Thank you,

Best Regards,
PAL Robotics Team

closed 2022-05-19 by saikishor