pal-robotics/aruco_ros #84: no aruco package for ros-melodic

seigot, 2020-07-11:
I tried to install ros-melodic-aruco-ros on ros-melodic, but there're no aruco package.

```
$ sudo apt-get install ros-melodic-aruco-ros
パッケージリストを読み込んでいます... 完了
依存関係ツリーを作成しています                
状態情報を読み取っています... 完了
E: パッケージ ros-melodic-aruco-ros が見つかりません
```

Do you have any plans to release for ros-melodic-aruco-ros?

ManciuCen, 2020-07-17:
I think you can just download the source and compile it. use 'catkin make install'

seigot, 2020-07-19:
thanks, I got installed.

```
cd catkin_src/src
git clone https://github.com/pal-robotics/aruco_ros
cd ..
catkin make install
```

Do you know how to release as an apt-get package?
I'd like to make the install process easier...

v-lopez, 2020-09-17:
Requested release in https://github.com/ros/rosdistro/pull/26584

closed 2020-09-17 by v-lopez