pal-robotics/aruco_ros #87: rostopic echo /aruco_simple/pose not working

Abhinavgandhi09, 2020-09-23:
Config: ROS 1, Melodic, Ubuntu 18.04

Hi,

I am trying to track 2 markers with the double.launch file. The 2 markers are detected and I can see the result using image_view node in the usb_cam pkg. However, when I echo the pose and pose2 topics they are blank.

I can echo the /tf topic and see the hand and object frames being published. Please let me know if you need additional info!

saikishor, 2022-04-01:
Hello @Abhinavgandhi09 ,

We are extremely sorry for the late response. Could you give it a shot with the new changes in the repository and get back to us?.

Thank you, 

Best Regards,
PAL Robotics Team.

Sun-Jing-Kang, 2022-07-13:
Hi,

I have the same problem! When I do 'rostopic echo /aruco_single/pose', it seems no message be sended:

[image]

When I do 'rostopic echo /aruco_single/result', I can see the result:

[image]

What should I do for this problem? My ros in kenitic, ubuntu16.04

Thanks!

saikishor, 2022-07-13:
Hello @Sun-Jing-Kang and @Abhinavgandhi09 

Thank you for your interest in the `aruco_ros` library. Regarding your issue, the aruco_ros single only publishes the info of the marker and its pose only for [the matching marker ID](https://github.com/pal-robotics/aruco_ros/blob/noetic-devel/aruco_ros/src/simple_single.cpp#L214-L272). The one you see in the image are all the detected ArUcos, but if the detected ArUco marker is of not the intended one, it doesn't publish the info. 

For instance, as below you might detect 100s of ArUcos but you might be interested in a specific one. You would need to specify the marker_id while launching the launch file with the argument: `markerId:=582`

[image]

We hope this solves your issue.

Thank you,

Best Regards,
PAL Robotics Team

closed 2022-07-13 by saikishor