pal-robotics/aruco_ros #89: No version for Noetic OpenCV4

RDaneelOlivav, 2020-12-09:
Hi,

Trying to compile for noetic OpenCV 4 but its crashing everywhere. An plan for a Noetic release? Thanks

v-lopez, 2020-12-14:
Not immediately, we may even invest time on ROS2 before noetic (which will have also OpenCV4, but will change a lot more code), but if anyone can submit a PR for noetic we'll take a look.

felixvd, 2021-02-10:
@4lhc Is there a reason you didn't make a PR for that branch? It compiled on Noetic for me.

4lhc, 2021-02-11:
@felixvd I didn't make a PR because there was an issue with the marker pose that I noticed, the cause of which I was not sure of. The marker pose was always at an offset from the real position of the marker as seen in the attached image. I didn't get time to explore it further. 

[image]

OTL, 2021-05-20:
@4lhc I've checked your code on noetic, and it seems that it works perfectly for me.
[image]

lucamarchionna, 2021-07-05:
Hi everyone, I was trying to use this repository on Noetic but I'm having problems when compiling this package, due to the wrong version of OpenCV. 
Snce you solved the problem, how can I fix this issue? Thanks in advance.

OTL, 2021-07-05:
I think https://github.com/4lhc/aruco_ros just works fine.

v-lopez, 2021-07-05:
@4lhc feel free to make a PR and if the issue you mention is verified we can fix it.

4lhc, 2021-07-05:
@v-lopez Sure, go ahead.

v-lopez, 2021-07-07:
Done!

closed 2021-07-07 by v-lopez