pal-robotics/aruco_ros #92: High CPU usage

xudooo, 2021-03-23:
Hello,
I'm using aruco to track a tag (640x480 30fps) and CPU load  goes up to 90% on my core i5-10500 CPU (3.10GHz×12),  ubuntu 18.04.5 LTS system. When it cannot detect a tag, the usage even stabilizes at 95%. Is it normal?  
I need to run another program at the same time. What should I do to reduce the CPU usage aruco? Thank you.
[image]

saikishor, 2021-03-23:
Hello @xudooo ,

Thanks for writing to us. I assume you were running the node and listening to the topics continuously. The CPU load doesn't depend on whether you have the ArUco marker present in the image or not, it still processes the image to look for the marker. 

If you see here https://github.com/pal-robotics/aruco_ros/blob/melodic-devel/aruco_ros/src/simple_single.cpp#L187-L190, it does process the image only when there are subscribers to the topics the node is publishing, else it won't process any image. 

We highly recommend you to only subscribe to these topics only when you need the information, in this way you could reduce the CPU usage with the node.

xudooo, 2021-03-24:
I'll consider your suggestion, thanks for reply.