pal-robotics/aruco_ros #98: Wrong pose of aruco tag after upgrade to ros noetic

hacmue, 2022-02-10:
We have migrated our software from ROS Melodic to ROS Noetic. I noticed that the pose from the Aruco tag is no longer correct. Also, the coordinate system in the result image is drawn in the wrong position (similar to https://github.com/pal-robotics/aruco_ros/issues/89#issuecomment-777649188) .

With version 2.0.2:
[image]

With commit 52c8c2bd4e9f60b6ff6a55e9470ba879fb5158ef : 
[image]

I use the following launch-file:
`<launch>

    <arg name="markerSize"      default="0.11"/>    
    <arg name="side"             default="front"/>
    <arg name="ref_frame"       default="camera_$(arg side)_depth_optical_frame"/>  

    <node pkg="aruco_ros" type="marker_publisher" name="aruco_marker_publisher_$(arg side)_camera">
        <remap from="/camera_info" to="/camera_$(arg side)/color/camera_info" />
        <remap from="/image" to="/camera_$(arg side)/color/image_raw" />
        <remap from="~/markers" to="/aruco/markers" />
        <param name="image_is_rectified" value="False"/>
        <param name="marker_size"        value="$(arg markerSize)"/>
        <param name="reference_frame"    value="$(arg ref_frame)"/>   
        <param name="camera_frame"       value="camera_$(arg side)_depth_optical_frame"/>
    </node>

</launch>
```
The camera_info-message
`rostopic echo -n 1 /camera_front/color/camera_info 
header: 
  seq: 13528
  stamp: 
    secs: 4155
    nsecs: 142000000
  frame_id: "camera_front_depth_optical_frame"
height: 480
width: 640
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [319.9988245765257, 0.0, 320.5, 0.0, 319.9988245765257, 240.5, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [319.9988245765257, 0.0, 320.5, -0.0, 0.0, 319.9988245765257, 240.5, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
```

saikishor, 2022-02-10:
Hello @hacmue ,

Thank you for showing interest in the package. Could you please test the same by passing a rectified image and by setting the param `image_is_rectified` to True, let us know if it fixes it?, why is your camera frame pointing to the `camera_depth_optical_frame` instead of the `camera_color_optical_frame`?.

It seems using the rectified image this doesn't happen:
[image]

However, When using a non rectified image in the simulation, we could reproduce the issue you were mentioning
[image]

We would try to fix it in the upcoming weeks after some internal planning, however, if you can open a PR, we can take a look.

Thank you,

Best Regards,
PAL Robotics Team

saikishor, 2022-02-10:
Hello @hacmue,

We have found the bug and fixed it. Probably, tomorrow or next week you will have fix here in GitHub. Thank you for reporting the issue.

Best Regards,
PAL Robotics Team.

saikishor, 2022-02-10:
Hello @hacmue,

We have just updated our repository with the fix. We will be closing this issue as the fix has been deployed. We thank you again for pointing out the issue. In case of queries, feel free to open the issue again.

Thank you,

Best Regards,
PAL Robotics Team.

closed 2022-02-10 by saikishor

hacmue, 2022-02-11:
Thanks a lot. It works fine.