pal-robotics/aruco_ros #100: Aruco markers not detected in Gazebo

Mechaick, 2022-03-16:
Hello,

I'm currently running the following gazebo simulation :
https://github.com/EuropeanRoverChallenge/ERC-Remote-Navigation-Sim#launching

My issue is that aruco_single and aruco_marker_publisher only sees the markers for a frame then stops seeing them. Do you have any idea why ?

This is my launch file

`<launch>

    <arg name="markerSize"      default="0.05"/>    
    <arg name="ref_frame"       default="base"/>  
    <arg name="use_camera_info"	default="true"/>

    <node pkg="aruco_ros" type="marker_publisher" name="aruco_marker_publisher">
        <remap from="/camera_info" to="/zed2/right/camera_info" />
        <remap from="/image" to="/zed2/right/image_rect_color" />
        <param name="image_is_rectified" value="True"/>
        <param name="marker_size"        value="$(arg markerSize)"/>
        <param name="reference_frame"    value="$(arg ref_frame)"/>   
        <param name="camera_frame"       value="zed2_right_camera_optical_frame"/>
    </node>

</launch>
`

These are my topics

`
/aruco_marker_publisher/debug
/aruco_marker_publisher/debug/compressed
/aruco_marker_publisher/debug/compressed/parameter_descriptions
/aruco_marker_publisher/debug/compressed/parameter_updates
/aruco_marker_publisher/debug/compressedDepth
/aruco_marker_publisher/debug/compressedDepth/parameter_descriptions
/aruco_marker_publisher/debug/compressedDepth/parameter_updates
/aruco_marker_publisher/debug/theora
/aruco_marker_publisher/debug/theora/parameter_descriptions
/aruco_marker_publisher/debug/theora/parameter_updates
/aruco_marker_publisher/markers
/aruco_marker_publisher/markers_list
/aruco_marker_publisher/result
/aruco_marker_publisher/result/compressed
/aruco_marker_publisher/result/compressed/parameter_descriptions
/aruco_marker_publisher/result/compressed/parameter_updates
/aruco_marker_publisher/result/compressedDepth
/aruco_marker_publisher/result/compressedDepth/parameter_descriptions
/aruco_marker_publisher/result/compressedDepth/parameter_updates
/aruco_marker_publisher/result/theora
/aruco_marker_publisher/result/theora/parameter_descriptions
/aruco_marker_publisher/result/theora/parameter_updates
/camera/camera_info
/camera/image_raw
/camera/image_raw/compressed
/camera/image_raw/compressed/parameter_descriptions
/camera/image_raw/compressed/parameter_updates
/camera/image_raw/compressedDepth
/camera/image_raw/compressedDepth/parameter_descriptions
/camera/image_raw/compressedDepth/parameter_updates
/camera/image_raw/theora
/camera/image_raw/theora/parameter_descriptions
/camera/image_raw/theora/parameter_updates
/camera/parameter_descriptions
/camera/parameter_updates
/clock
/cmd_vel
/gazebo/controllers/diff_drive/cmd_vel
/gazebo/controllers/diff_drive/odom
/gazebo/controllers/diff_drive/parameter_descriptions
/gazebo/controllers/diff_drive/parameter_updates
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/performance_metrics
/gazebo/set_link_state
/gazebo/set_model_state
/gazebo/zed2/odom
/gazebo/zed2/odom_info
/gazebo/zed2/odom_info_lite
/gazebo/zed2/odom_last_frame
/gazebo/zed2/odom_local_map
/gazebo/zed2/odom_local_scan_map
/gazebo/zed2/odom_rgbd_image
/gazebo_ros_control/pid_gains/wheel_FL_joint/parameter_descriptions
/gazebo_ros_control/pid_gains/wheel_FL_joint/parameter_updates
/gazebo_ros_control/pid_gains/wheel_FR_joint/parameter_descriptions
/gazebo_ros_control/pid_gains/wheel_FR_joint/parameter_updates
/gazebo_ros_control/pid_gains/wheel_RL_joint/parameter_descriptions
/gazebo_ros_control/pid_gains/wheel_RL_joint/parameter_updates
/gazebo_ros_control/pid_gains/wheel_RR_joint/parameter_descriptions
/gazebo_ros_control/pid_gains/wheel_RR_joint/parameter_updates
/ground_truth
/joint_states
/probe_deployment_unit/drop
/probe_deployment_unit/probes_dropped
/rosout
/rosout_agg
/tf
/tf_static
/wheel_odom
/wheel_pose
/zed2/depth/camera_info
/zed2/depth/depth_registered
/zed2/imu/data
/zed2/left/camera_info
/zed2/left/image_rect_color
/zed2/left/image_rect_color/compressed
/zed2/left/image_rect_color/compressed/parameter_descriptions
/zed2/left/image_rect_color/compressed/parameter_updates
/zed2/left/image_rect_color/compressedDepth
/zed2/left/image_rect_color/compressedDepth/parameter_descriptions
/zed2/left/image_rect_color/compressedDepth/parameter_updates
/zed2/left/image_rect_color/theora
/zed2/left/image_rect_color/theora/parameter_descriptions
/zed2/left/image_rect_color/theora/parameter_updates
/zed2/left_cam_imu_transform
/zed2/left_raw/camera_info
/zed2/left_raw/image_raw_color
/zed2/left_raw/image_raw_color/compressed
/zed2/left_raw/image_raw_color/compressed/parameter_descriptions
/zed2/left_raw/image_raw_color/compressed/parameter_updates
/zed2/left_raw/image_raw_color/compressedDepth
/zed2/left_raw/image_raw_color/compressedDepth/parameter_descriptions
/zed2/left_raw/image_raw_color/compressedDepth/parameter_updates
/zed2/left_raw/image_raw_color/theora
/zed2/left_raw/image_raw_color/theora/parameter_descriptions
/zed2/left_raw/image_raw_color/theora/parameter_updates
/zed2/left_raw/parameter_descriptions
/zed2/left_raw/parameter_updates
/zed2/odom
/zed2/parameter_descriptions
/zed2/parameter_updates
/zed2/point_cloud/cloud_registered
/zed2/reset_odometry
/zed2/right/camera_info
/zed2/right/image_rect_color
/zed2/right/image_rect_color/compressed
/zed2/right/image_rect_color/compressed/parameter_descriptions
/zed2/right/image_rect_color/compressed/parameter_updates
/zed2/right/image_rect_color/compressedDepth
/zed2/right/image_rect_color/compressedDepth/parameter_descriptions
/zed2/right/image_rect_color/compressedDepth/parameter_updates
/zed2/right/image_rect_color/theora
/zed2/right/image_rect_color/theora/parameter_descriptions
/zed2/right/image_rect_color/theora/parameter_updates
/zed2/right/parameter_descriptions
/zed2/right/parameter_updates
/zed2/right_raw/camera_info
/zed2/right_raw/image_raw_color
/zed2/right_raw/image_raw_color/compressed
/zed2/right_raw/image_raw_color/compressed/parameter_descriptions
/zed2/right_raw/image_raw_color/compressed/parameter_updates
/zed2/right_raw/image_raw_color/compressedDepth
/zed2/right_raw/image_raw_color/compressedDepth/parameter_descriptions
/zed2/right_raw/image_raw_color/compressedDepth/parameter_updates
/zed2/right_raw/image_raw_color/theora
/zed2/right_raw/image_raw_color/theora/parameter_descriptions
/zed2/right_raw/image_raw_color/theora/parameter_updates
/zed2/right_raw/parameter_descriptions
/zed2/right_raw/parameter_updates
`

and this is the command I use : 
`roslaunch aruco_ros marker_publisher.launch markerSize:=0.15 ref_frame:=/base_link`

saikishor, 2022-04-01:
Hello @Mechaick,

This could probably mean that the aruco_ros is not properly detecting the marker. Once it is able to detect the marker, then the pose estimation is done. 

Thank you,

Best Regards,
PAL Robotics Team.

saikishor, 2022-05-16:
Hello @Mechaick,

Any update on this issue?.

Thank you,

Best Regards,
PAL Robotics Team.

saikishor, 2022-11-15:
Hello @Mechaick,

Closing the issue due to inactivity.

Thank you,

Best Regards,
PAL Robotics Team.

closed 2022-11-15 by saikishor