pal-robotics/aruco_ros #105: Unable to build package

AngeloEspinoza, 2022-05-15:
Hi, I am currently trying to build the package however every time I try to do it I get

`[ 38%] Building CXX object aruco_ros/aruco_ros/CMakeFiles/marker_publisher.dir/src/marker_publish.cpp.o
/home/angelo/catkin_ws/src/aruco_ros/aruco_ros/src/marker_publish.cpp:45:10: fatal error: aruco_msgs/MarkerArray.h: No such file or directory
   45 | #include <aruco_msgs/MarkerArray.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [aruco_ros/aruco_ros/CMakeFiles/marker_publisher.dir/build.make:63: aruco_ros/aruco_ros/CMakeFiles/marker_publisher.dir/src/marker_publish.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:5139: aruco_ros/aruco_ros/CMakeFiles/marker_publisher.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j8 -l8" failed
`
I am not sure what it is, I've seen a similar issue that says the problem was the person was not building directly with `catkin_make`; however, I'm building in the workspace with `catkin_make`.

Thank you in advance.

saikishor, 2022-05-16:
Hello @AngeloEspinoza,

Could you pull the new changes in the repository and compile the package. Let us know if this fixes the issue for you.

Thank you,

Best Regards,
PAL Robotics Team.

saikishor, 2022-05-20:
Hello @AngeloEspinoza,

Were you able to solve the issue at your end?.

Best Regards,
PAL Robotics Team.

AngeloEspinoza, 2022-05-20:
Hi!

Still unable to build my workspace, I got:

`collect2: error: ld returned 1 exit status
make[2]: *** [aruco_ros/aruco_ros/CMakeFiles/single.dir/build.make:211: /home/angelo/catkin_ws/devel/lib/aruco_ros/single] Error 1
make[1]: *** [CMakeFiles/Makefile2:5235: aruco_ros/aruco_ros/CMakeFiles/single.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j8 -l8" failed
`

Regards,
Angelo

saikishor, 2022-05-20:
Hello @AngeloEspinoza ,

Thanks for the prompt reply. Unfortunately, the log you shared is not enough for us to debug, we couldn't reproduce the issue at our end. Could you clean your workspace and then run the `catkin build`. Let us know the output.

Thank you,

Best Regards,
PAL Robotics Team.

saikishor, 2022-05-20:
@AngeloEspinoza ,

If you are using melodic, please don't forget to switch the branch to `melodic-devel`.

Best Regards,
PAL Robotics Team.

AngeloEspinoza, 2022-05-20:
I'm using noetic distro. 
I just pull the repo into another workspace (I have one for `catkin_make` only and other for `catkin build`). However, still got the same issue. 
`collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/single.dir/build.make:196: /home/angelo/catkin_ws2/devel/.private/aruco_ros/lib/aruco_ros/single] Error 1
make[1]: *** [CMakeFiles/Makefile2:324: CMakeFiles/single.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/angelo/catkin_ws2/build/aruco_ros; catkin build --get-env aruco_ros | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

..................................................................................................................................................................................................................
Failed     << aruco_ros:make               [ Exited with code 2 ]                                                                                                                                                 
Failed    <<< aruco_ros                    [ 13.3 seconds ]                                                                                                                                                       
[build] Summary: 10 of 11 packages succeeded.                                                                                                                                                                     
[build]   Ignored:   None.                                                                                                                                                                                        
[build]   Warnings:  1 packages succeeded with warnings.                                                                                                                                                          
[build]   Abandoned: None.                                                                                                                                                                                        
[build]   Failed:    1 packages failed.                                                                                                                                                                           
[build] Runtime: 25.6 seconds total.                                                                                                                                                                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.
`

saikishor, 2022-05-20:
Hello @AngeloEspinoza ,

Can you please share the complete log?. It is hard for us to check the issue. We are still not able to reproduce the issue at our end, so we would need the whole log to check the issue. In the meantime, you could install the aruco_ros from the upstream using apt.

Thank you,

Best Regards,
PAL Robotics Team

AngeloEspinoza, 2022-05-20:
All right

```
-------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/angelo/catkin_ws/devel:/opt/ros/noetic
Workspace:                   /home/angelo/catkin_ws2
-------------------------------------------------------------------------
Build Space:        [exists] /home/angelo/catkin_ws2/build
Devel Space:        [exists] /home/angelo/catkin_ws2/devel
Install Space:      [unused] /home/angelo/catkin_ws2/install
Log Space:          [exists] /home/angelo/catkin_ws2/logs
Source Space:       [exists] /home/angelo/catkin_ws2/src
DESTDIR:            [unused] None
-------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
-------------------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------------------
[build] Found '11' packages in 0.0 seconds.                                                                                                                                                                       
[build] Updating package table.                                                                                                                                                                                   
Starting  >>> aruco                                                                                                                                                                                               
Starting  >>> aruco_msgs                                                                                                                                                                                          
Starting  >>> iq_vision                                                                                                                                                                                           
Starting  >>> mavlink                                                                                                                                                                                             
Starting  >>> mavros_msgs                                                                                                                                                                                         
Finished  <<< iq_vision                    [ 0.3 seconds ]                                                                                                                                                        
Finished  <<< mavlink                      [ 1.5 seconds ]                                                                                                                                                        
Starting  >>> libmavconn                                                                                                                                                                                          
Finished  <<< mavros_msgs                  [ 2.4 seconds ]                                                                                                                                                        
Finished  <<< libmavconn                   [ 0.2 seconds ]                                                                                                                                                        
Starting  >>> mavros                                                                                                                                                                                              
Finished  <<< aruco_msgs                   [ 6.7 seconds ]                                                                                                                                                        
Finished  <<< mavros                       [ 1.6 seconds ]                                                                                                                                                        
Starting  >>> iq_sim                                                                                                                                                                                              
Starting  >>> mavros_extras                                                                                                                                                                                       
Finished  <<< iq_sim                       [ 0.2 seconds ]                                                                                                                                                        
Finished  <<< mavros_extras                [ 0.9 seconds ]                                                                                                                                                        
[... cut ...]
[build]   Failed:    1 packages failed.                                                                                                                                                                           
[build] Runtime: 25.6 seconds total.                                                                                                                                                                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

closed 2022-05-20 by saikishor