pal-robotics/aruco_ros #106: Support of aruco for ROS2 is available?

goyalvarun02, 2022-09-20:
Hi ,

May I know if github of aruco for ROS 2 is available  Please share the path .

or if not available May I know when can expect the ros2 of aruco

Thanks

saikishor, 2022-09-21:
Hello!

We have started migrating our robots simulation to ROS2 and you will be able to find them online soon. However, the aruco_ros package is currently not in our near future roadmap. We would be happy to review and accept any PRs on ROS2.

Thank you,

Best Regards,
PAL Robotics Team.

YuLiHN, 2022-11-02:
Hey may I know where are you right now? Maybe I can help.

> Hello!
> 
> We have started migrating our robots simulation to ROS2 and you will be able to find them online soon. However, the aruco_ros package is currently not in our near future roadmap. We would be happy to review and accept any PRs on ROS2.
> 
> Thank you,
> 
> Best Regards, PAL Robotics Team.

saikishor, 2022-11-10:
Hello @YuLiHN,

Thank you for showing interest in the package. We are working on it, may be in next few weeks you can expect a version of ROS2.

Thank you,

Best Regards,
PAL Robotics Team.

saikishor, 2022-11-15:
Hello @goyalvarun02 and @YuLiHN,

We have just released the first version of ROS2 support for the aruco_ros package. You could use it by cloning and switching to the `humble-devel` branch. Let us know once you test it. 

Thank you,

Best Regards,
PAL Robotics Team.

closed 2022-11-15 by saikishor