pal-robotics/aruco_ros #107: Can't detect marker with realsense

0smile, 2022-09-27:
Hello all, 
I'm using a realsense D435i for acuro calibration. The odd thing is that it can't detect any acuro marker, so it is not a problem of a specific marker. In the image view of rqt, the topic of "/aruco_tracker/debug" shows thresholdied images with severe twinkle, as shown below:
[image]
[image]
[image]
Here is my launch file:
`<launch>
    <arg name="namespace_prefix" default="kinova_d435i_handeyecalibration" />
 
    <arg name="marker_size" value="0.1" />
    <arg name="marker_id" value="102" />
 

    <include file="$(find realsense2_camera)/launch/rs_camera.launch" >
    </include>
 
    
    <node name="aruco_tracker" pkg="aruco_ros" type="single">
        <remap from="/camera_info" to="/camera/color/camera_info" />
        <remap from="/image" to="/camera/color/image_raw" />
        <param name="image_is_rectified" value="true"/>
        <param name="marker_size"        value="$(arg marker_size)"/>
        <param name="marker_id"          value="$(arg marker_id)"/>
        <param name="reference_frame"    value="camera_link"/>
        <param name="camera_frame"       value="camera_depth_optical_frame"/>
        <param name="marker_frame"       value="camera_marker" />
 
    </node>

 
</launch>`
Wondering whether it is a problem of realsense or aruco_ros? Thanks for any help!

0smile, 2022-09-27:
I assume it is related to adaptive thresh method, if so how should I edit it into a fixed thresh method?

saikishor, 2022-09-27:
Hello @0smile,

Are you using the ArUco marker same as the ID you have specified in the launch file?

Best Regards,
PAL Robotics Team.

0smile, 2022-09-27:
Thank you @saikishor for replying! 
Yes, I am using the 102 marker. I generate it in https://chev.me/arucogen/, you can see it is the same as the one in the image viewer.
[image]

saikishor, 2022-09-27:
Hello @0smile,

We have used aruco_ros with the realsense in the past successfully. We don't think it's with the realsense itself. Could you test the same in the Gazebo and see if the setup works for you? or with a different camera if you believe it is from the realsense?

Best Regards,
PAL Robotics Team.

0smile, 2022-09-28:
Hello @saikishor,

I have tried another camera, still can't detect the marker. However, I found a similar issue on ros answer:
[https://answers.ros.org/question/388755/aruco_ros-notetecting-marker/](https://answers.ros.org/question/388755/aruco_ros-notetecting-marker/)
I have the same issue as:
`[ WARN] : normalizeImageIllumination is unimplemented!`
I think there might be issue with the normalization of image illumination, thus the thresholded image is unstable. To clarify, I'm using ROS noetic and the noetic branch for aruco_ros.

0smile, 2022-09-28:
After some searching I found it is a silly question. The reason is that when use https://chev.me/arucogen/, it offer aruco markers of 4X4 dictionary in default, but the aruco_ros seems use original dictionary in default. I will close the issue, sorry for interrupting!

closed 2022-09-28 by 0smile

saikishor, 2022-09-28:
Hello @0smile,

We are glad that you were able to figure out the issue with the AruCo generation.

Best Regards,
PAL Robotics Team.