pal-robotics/aruco_ros #108: cv::Exception OpenCV(4.2.0) 9004:Unknown status code 9004) !isValid(): invalid marker

Assets01, 2022-11-20:
Hi, 
I am using the noetic-devel branch on ubuntu 20.04 and ros noetic, but an opencv error occurred when I launch the single.launch with edited params for intel realsense d435, like this:

terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) /home/wyx/catkin_ws/src/aruco_ros/aruco/src/aruco/marker.cpp:321: error: (9004:Unknown status code 9004) !isValid(): invalid marker. It is not possible to calculate extrinsics in function 'calculateExtrinsics'

Assets01, 2022-11-20:
any one konws how to solve this?

Assets01, 2022-11-20:
the launch file is:

<launch>

    <arg name="markerId"        default="582"/>
    <arg name="markerSize"      default="0.05"/>    
    <arg name="eye"             default="left"/>
    <arg name="marker_frame"    default="aruco_marker_frame"/>
    <arg name="ref_frame"       default=""/>  
    <arg name="corner_refinement" default="LINES" /> 

    <node pkg="aruco_ros" type="single" name="aruco_single">
        <remap from="/camera_info" to="/camera/color/camera_info" />
        <remap from="/image" to="/camera/color/image_raw" />
        <param name="image_is_rectified" value="True"/>
        <param name="marker_size"        value="$(arg markerSize)"/>
        <param name="marker_id"          value="$(arg markerId)"/>
        <param name="reference_frame"    value="$(arg ref_frame)"/>   
        <param name="camera_frame"       value="camera_link"/>
        <param name="marker_frame"       value="$(arg marker_frame)" />
        <param name="corner_refinement"  value="$(arg corner_refinement)" />
    </node>

</launch>

Assets01, 2022-11-20:
and I tried on another PC with ubuntu 16.04 and ros kinetic, it works fine.

saikishor, 2022-11-20:
Hello @Assets01,

Could you please try the same by checking out the tag 3.0.3 and get back to us?.

`git checkout 3.0.3`

Thank you,

Best Regards,
PAL Robotics Team.

Assets01, 2022-11-20:
> 

Thanks, problem solved!

saikishor, 2022-11-20:
Hello @Assets01,

Awesome, so if we understand correctly the issue doesn't happen with tag `3.0.3`, but with the main branch you could reproduce it. Please let us know.

Thank you,

Best Regards,
PAL Robotics Team.

Assets01, 2022-11-20:
After I checking out the noetic-devel, the same error happens again. So I guess this error is related to the main branch. @saikishor

saikishor, 2022-11-20:
Hello @Assets01,

Thank you so much for confirming it. Could you make a small rosbag with the topics used by the aruco_ros and share with us, so we could reproduce the error and fix it at our end . In our initial tests, the node performed similar to earlier. This data that you might provide could help us in fixing the bug in newer version of aruco_ros.

Thank you,

Best Regards,
PAL Robotics Team.

joaopsousa99, 2022-11-21:
Hello, @saikishor ,

I am experiencing the same issue as @Assets01 and I have tried `git checkout 3.0.3` and it didn't change anything. I am also on the `noetic-devel` branc, but on ubuntu 20.04.

saikishor, 2022-11-21:
Hello @joaopsousa99,

Thank you for the information. You mean that the issue persists even with 3.0.3?. Could you please send us a small rosbag with all the topics subscribed + /tf, so we can debug from our end?. We have tried to reproduce this morning but unfortunately we couldn't. We need a rosbag with proper information.

Thank you,

Best Regards,
PAL Robotics Team.

saikishor, 2022-11-21:
Hello @joaopsousa99,

Could you add it to your Google Drive and share a temporary link here?

Thank you,

Best Regards,
PAL Robotics Team.

saikishor, 2022-11-23:
Hello @670768312,

We have fixed the bug you are facing. You can get the fixed by doing `git pull noetic-devel`  and compile.

Thank you,

Best Regards,
PAL Robotics Team.

closed 2022-11-23 by saikishor