pal-robotics/aruco_ros #109: aruco_ros/aruco/src/aruco/marker.cpp:321: error: (9004) !isValid(): invalid marker.

670768312, 2022-11-22:
Hi,
I am using the melodic-devel branch on ubuntu 18.04 and ros melodic,but when I run“ roslaunch aruco_ros single.launch”，when using rqt "/aruco_single/reslut"to test the effect and the camera model is D435i, an error will be reported,like this:
"terminate called after throwing an instance of 'cv::Exception'
  what():  /home/heart/catkin_ws/src/aruco_ros/aruco/src/aruco/marker.cpp:321: error: (9004) !isValid(): invalid marker. It is not possible to calculate extrinsics in function calculateExtrinsics

[aruco_single-1] process has died [pid 21424, exit code -6, cmd /home/heart/catkin_ws/devel/lib/aruco_ros/single /camera_info:=/camera/color/camera_info /image:=/camera/color/image_raw __name:=aruco_single __log:=/home/heart/.ros/log/0f22ed66-6a44-11ed-aa5f-e45e3755f91a/aruco_single-1.log].
log file: /home/heart/.ros/log/0f22ed66-6a44-11ed-aa5f-e45e3755f91a/aruco_single-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done"
Thank you very much for helping me solve this problem！！！

saikishor, 2022-11-22:
Hello @670768312,

Thank you for pointing out the issue. Could you please test the same with the tag  ` 2.1.4`. You can do that simply by  `git checkout 2.1.4` . Please kindly let us know if you face this issue with this particular tag. If this is not the case, we kindly request you to share a small bag to reproduce the bug along with the /tf topic.

Thank you,

Best Regards,
PAL Robotics Team.

670768312, 2022-11-22:
Thank you very much ! I have solved the current problem according to your method, thank you again, love you！！！

saikishor, 2022-11-22:
Hello @670768312,

Could you please provide a rosbag with the topics and the /tf topic so we can try to reproduce the error and try to fix it. You could upload it to your drive and share a temporary link here. 

Thank you,

Best Regards,
PAL Robotics Team.

670768312, 2022-11-22:
Sorry,I have just contacted the ROS system, and I don't understand what you need me to do

saikishor, 2022-11-22:
Hello @670768312,

It would be interesting to share your roslaunch info too. 

Thanks,

Best Regards,
PAL Robotics Team.

saikishor, 2022-11-22:
Hello @670768312,

> Sorry,I have just contacted the ROS system, and I don't understand what you need me to do

You can record some topics using rosbag interface. Check this link below for more info : http://wiki.ros.org/rosbag/Commandline#record

Best Regards,
PAL Robotics Team

670768312, 2022-11-22:
I use the command“git clone -b melodic-devel https://github.com/pal-robotics/aruco_ros.git” to download the package. After compiling, I use D435i to test and rqt to view the results. The error just described will appear.

saikishor, 2022-11-22:
Hello @670768312,

We tried to reproduce the error in a simulation but we couldn't. We will try to test it with D435i and check it.

Thank you,

Best Regards,
PAL Robotics Team.

670768312, 2022-11-22:
OK,

Thanks again for your help

saikishor, 2022-11-23:
Hello @670768312,

We have fixed the bug you are facing. You can get the fixed by doing `git pull melodic-devel`  and compile.

Thank you,

Best Regards,
PAL Robotics Team.