pal-robotics/aruco_ros #110: aruco_ros on noetic only detects AprilTag and original ArUco markers

joaopsousa99, 2022-11-25:
I'm running `aruco_ros` with the command `roslaunch aruco_ros single.launch markerId:=0` and testing the detection with a marker from several different dictionaries available in OpenCV (Python 3's `cv2`), all with ID 0. The dictionaries I am using are:
- `cv2.aruco.DICT_4X4_50`
- `cv2.aruco.DICT_5X5_50`
- `cv2.aruco.DICT_6X6_50`
- `cv2.aruco.DICT_7X7_50`
- `cv2.aruco.DICT_ORIGINAL`
- `cv2.aruco.DICT_APRILTAG_25h9`
- `cv2.aruco.DICT_APRILTAG_36h10`
- `cv2.aruco.DICT_APRILTAG_36h11`

All the AprilTag and the ArUco Original dictionaries are being detected just fine. The others are not detected at all.

After some research, I thought the problem was due to me having generated all the `cv2.aruco.DICT[4:7]X[4:7]` with the `_50` at the end, which, according to [this](https://docs.opencv.org/3.4/d9/d6a/group__aruco.html#gac84398a9ed9dd01306592dd616c2c975), corresponds to a minimum hamming distance of 4, 8, 13 and 19, respectively. I tried using `cv2.aruco.DICT_4X4_250`, `cv2.aruco.DICT_5X5_100` and `cv2.aruco.DICT_6X6_250` instead (no `7X7` option available for `aruco_ros`), which, according to [this](http://boofcv.org/javadoc/boofcv/factory/fiducial/HammingDictionary.html), correspond to `aruco_ros`'s `ARUCO_MIP_16h3`, `ARUCO_MIP_25h7` and `ARUCO_MIP_36h12`.

However, when I looked at the resulting markers, they are equal to the ones I had already generated (the ones from the dictionaries ending in `_50`).

This leads me to think the problem lies on the package and not the markers.

OS: Ubuntu 20.04 LTS
ROS: Noetic

saikishor, 2022-11-28:
Hello @joaopsousa99,

What's the problem exactly?. As per the title, the package does detect ArUco markers and we don't see the issue here. Please try to post info where we can reproduce it at our end.

Thank you,

Best Regards,
PAL Robotics Team.

joaopsousa99, 2022-11-28:
Sorry, I'll try to be clearer.

The issue is that, out of all the dictionaries I listed, the only only detects:
- `cv2.aruco.DICT_ORIGINAL`
- `cv2.aruco.DICT_APRILTAG_25h9`
- `cv2.aruco.DICT_APRILTAG_36h10`
- `cv2.aruco.DICT_APRILTAG_36h11`

To reproduce the error:
1. Start camera drivers
2. `roslaunch aruco_ros single.launch markerId:=0`
3. Point camera at a marker from either the `4X4`, the `5X5`, the `6X6` or the `7X7` dictionaries
4. Point camera at a marker from either the `DICT_ORIGINAL` or one of the AprilTag dictionaries
5. See that in step 4 nothing is detected
6. See that in step 5 any marker is correctly detected

saikishor, 2022-11-28:
Hello @joaopsousa99,

This library only works with the Original ArUco markers. It doesn't support other marker types. They are completely different information encoded into them. You can generate your tag using : [https://chev.me/arucogen/](https://chev.me/arucogen/) choosing Original ArUco in the drop down. Moreover, conceptually it doesn't make sense for us to support different marker dictionaries, imagine you are interested in getting the information of Marker 0, and then you find 3-4 markers of different types in the scene, you don't want the pose estimation to jump from one marker to another. 

Thank you,

Best Regards,
PAL Robotics Team.

closed 2022-11-28 by saikishor

joaopsousa99, 2022-11-28:
I see, I thought the package worked with all ArUco dictionaries available in OpenCV. Thank you for you help.

As a sidenote: why does the package support AprilTag, then?

saikishor, 2022-11-28:
Hello @joaopsousa99,

> As a sidenote: why does the package support AprilTag, then?
Regarding this we are unsure, we can dig into this in future. Thanks for bringing it to our notice.

Best Regards,
PAL Robotics Team.