pal-robotics/aruco_ros #111: No marker TF frame

saprrow, 2022-12-25:
I launched an aruco_ros single node and the marker could be recognized. But it's wired that no marker TF frame but realsense_camera frames can be found in the tf_frame.

Hope for a solution. Thanks!

[image]
[image]
![Screenshot from 2022-1
[image]
2-25 20-54-24](https://user-images.githubusercontent.com/44662300/209469336-bd2864db-2fae-4bdf-b434-bd9ea52c888a.png)

AminSeffo, 2023-02-25:
Hey @saprrow,
I am facing exactly the same problem, did you manage to solve this and can you tell me how?
Thanks in advance

saprrow, 2023-03-03:
Hi, AminSeffo
I just turned to easy_aruco and it works. Remeber to remap you camera topic to easy_aruco's default camera topic, or you can just change dafault topic name in 'src/Detector.cpp'.

saikishor, 2023-03-03:
Hello @saprrow and @AminSeffo,

I apologise for the delay in responding to this issue. Regarding your issues, arucos_ros only publishes the TF, where someone subscribes to the topics published by the node as specified in this part of the code: https://github.com/pal-robotics/aruco_ros/blob/noetic-devel/aruco_ros/src/simple_single.cpp#L187-L194. You can add a subscriber to one of these topics : https://github.com/pal-robotics/aruco_ros/blob/noetic-devel/aruco_ros/src/simple_single.cpp#L131-L137 to later enable the TF broadcasting. It is done this way to not to run the aruco detection all the time and thereby to reduce the load on the PC. We hope this helps to solve your issue.

Thank you,

Best Regards,
PAL Robotics Team.

closed 2024-02-18 by saprrow