pal-robotics/aruco_ros #112: Markers detected but no pose data

VGSTU, 2023-02-21:
Hello, I'm working on a project using Tello drones to detect aruco markers,
the makers can be detected and the id is recognized,
however the contents in published pose data are all 'nan'.
I'm using ROS melodic.

The launch file is as following
```
<launch>
    <!-- modification of "double.launch"-->
    <arg name="marker1Id"         default="582"/>  <!--set your maker id numbers-->
    <arg name="marker2Id"         default="26"/>
    <arg name="marker3Id"         default="100"/>
    <arg name="marker4Id"         default="70"/>
    <arg name="marker5Id"         default="55"/> 
    <arg name="markerSize"        default="0.07"/>  <!-- in m, use 7cm x 7cm markers -->
    <arg name="eye"               default="left"/>

    <arg name="dct_normalization" default="False" />
    <arg name="dct_filter_size"   default="2" /> 
    <arg name="marker1_frame"     default="tello_marker1_frame" />
    <arg name="marker2_frame"     default="tello_marker2_frame" />
    <arg name="marker3_frame"     default="tello_marker3_frame" />
    <arg name="marker4_frame"     default="tello_marker4_frame" />
    <arg name="marker5_frame"     default="tello_marker5_frame" />

    <node pkg="aruco_ros" type="penta" name="aruco_simple">
        <remap from="/camera_info" to="/tello/camera/camera_info" />
        <remap from="/image" to="/tello/image_raw"/>
        
        <param name="image_is_rectified" value="True"/>
        <param name="marker_size" value="$(arg markerSize)"/>
        <param name="marker_id1" value="$(arg marker1Id)"/>
        <param name="marker_id2" value="$(arg marker2Id)"/>
        <param name="marker_id3" value="$(arg marker3Id)"/>
        <param name="marker_id4" value="$(arg marker4Id)"/>
        <param name="marker_id5" value="$(arg marker5Id)"/>
        <param name="normalizeImage" value="$(arg dct_normalization)"/>
        <param name="dct_components_to_remove" value="$(arg dct_filter_size)"/>
        <param name="parent_name" value="tello_camera_optical_frame"/>
        <param name="child_name1" value="$(arg marker1_frame)" />
        <param name="child_name2" value="$(arg marker2_frame)" />
        <param name="child_name3" value="$(arg marker3_frame)" />
        <param name="child_name4" value="$(arg marker4_frame)" />
        <param name="child_name5" value="$(arg marker5_frame)" />
    </node>

</launch>
```

VGSTU, 2023-02-22:
It turned out the problem is caused by wrong camera calibration parameters.
Solved by re-calibrating and correcting the parameter path.
The issue will be closed.

closed 2023-02-22 by VGSTU

saikishor, 2023-02-22:
Hello @VGSTU,

We are happy that you solved the issue. We will try to add a concrete error to specify it at the runtime.

Thank you,

Best Regards,
PAL Robotics Team.