pal-robotics/aruco_ros #113: Memory errors for branch aruco_ros/tree/humble-devel

hizam-otsaw, 2023-04-05:
Hi,

I am trying out the ROS2 branch `aruco_ros/tree/humble-devel` and encountered the following memory errors. Strangely, these memory errors only occur when I try to tried to echo any of the output topics (e.g. `/aruco_single/result` or `/aruco_single/pose`

```
[single-1] terminate called after throwing an instance of 'cv::Exception'
[single-1]   what():  OpenCV(4.6.0) /root/opencv/modules/core/src/matrix.cpp:246: error: (-215:Assertion failed) s >= 0 in function 'setSize'
[single-1]

[single-1] terminate called after throwing an instance of 'cv::Exception'
[single-1]   what():  OpenCV(4.6.0) /root/opencv/modules/core/src/alloc.cpp:73: error: (-4:Insufficient memory) Failed to allocate 1318115504640 bytes in function 'OutOfMemoryError'
[single-1] 
```

I am inputting Realsense cam feed to it and the ROS 1 `aruco_ros/tree/melodic-devel` was previously working on my realsense. Any ideas on how I can resolve this? Thank you

awesomebytes, 2023-04-05:
That looks like it's trying to allocate an absurd amount of bytes (1.31811550464 Terabytes)!
To help the guys at PAL, if you have a small rosbag that can help trigger this issue, I'm sure they'd appreciate it.

>  Strangely, these memory errors only occur when I tried to echo any of the output topics 
This is normal as the topics are lazily computed+published. If there's no subscriber, it won't do anything to save CPU. ( https://github.com/pal-robotics/aruco_ros/blob/humble-devel/aruco_ros/src/simple_single.cpp#L227-L233 )

hizam-otsaw, 2023-04-05:
Thanks!

I have uploaded the following files in the link below:

a) 4 mins rosbag of the 2 camera input topics (`/color/camera_info` and `/color/image_raw`)
b) single.launch.py where I remapped the 2 camera input topics

Link: https://drive.google.com/drive/folders/1Dcq_vPu1AYuMYT56P4T5vmTOy1t97KXL?usp=share_link

saikishor, 2023-04-05:
Hello @hizam-otsaw,

Thank you for reporting the issue. We will try to reproduce the issue in the coming days and we will get back to you. 
@awesomebytes Thank you for requesting the bag in the first place.

Thank you,

Best Regards,
PAL Robotics Team.

saikishor, 2023-04-05:
Hello @hizam-otsaw,

We are lacking permission to access the files in the Google Drive. I've requested access to them, when you can please grant us access.

Thank you, 

Best Regards,
PAL Robotics Team.

hizam-otsaw, 2023-04-05:
Sorry @saikishor, I just granted the Google Drive access

saikishor, 2023-04-05:
Hello @hizam-otsaw,

We don't find the rosbag in the drive, we only find the metadata.yaml in the folder `rosbag2_camera`. Could you please verify and let us know.

Thank you,

Best Regards,
PAL Robotics Team.

hizam-otsaw, 2023-04-05:
Hi @saikishor,

I have re-uploaded the rosbag in the drive. You should be able to see it now. Thank you.

saikishor, 2023-04-05:
Hello @hizam-otsaw,

We have received the bag. However, We aren't able to reproduce the problem. We think the bag you passed to us is not helpful, as there are no ArUco markers pointed at the camera in the rosbag, and moreover, 80% of the bag that the camera points only to this scene:
 
[image]

We played the bag and then launched `ros2 launch aruco_ros single.launch.py` with the changes that you have shared with us. We could see in the `/aruco_single/debug` that the `aruco_single`  node is processing the images correctly.

Please share with us the proper instructions and valid data to reproduce the issue.

Thank you,

Best Regards,
PAL Robotics Team.

hizam-otsaw, 2023-04-05:
Hi @saikishor,

Thanks for validating the rosbag. Actually, I could not even process the image correctly with no aruco marker in it where the node will throw out memory errors when I start to publish one of the `/aruco_single` topics. 

I am running the node in a docker container which suggests that there are possibly some issues with my environment. I will try to test the package natively on a host machine and will update if I get such memory errors.

In the meantime, I have also uploaded another rosbag in the Google Drive link with the following image (aruco_marker_ID=200) which also throw out the memory errors.

[image]

Thank you.

saikishor, 2023-04-05:
Hello @hizam-otsaw,

We have just tested with the new bag file and even played in a loop for more than 5 min, we are unable to reproduce the crash due to insufficient memory. We are also using a docker to reproduce the issue. 

[image]

I'm also able to get the estimation of the aruco marker as below:
```---
header:
  stamp:
    sec: 1680690459
    nanosec: 747473408
  frame_id: stereo_gazebo_left_camera_optical_frame
pose:
  position:
    x: -0.01364164799451828
    y: 0.005413014441728592
    z: 0.11268073320388794
  orientation:
    x: 0.9991355241020884
    y: 0.01801236597650246
    z: -0.028866679870773817
    w: -0.02388459634182615
---
```

Thank you,

Best Regards,
PAL Robotics Team.

hizam-otsaw, 2023-04-06:
Hi @saikishor,

Thanks for the reply. I have found the cause and it is due to my openCV library version being different (4.6.0).

I have used ROS2 Humble's default openCV library (4.5.0) and no longer encounter the memory errors. Thanks for helping me with this issue!

saikishor, 2023-04-06:
Hello @hizam-otsaw,

We are glad you figured it out and fixed the issue on your end.

Thank you,

Best Regards,
PAL Robotics Team.

closed 2023-04-06 by saikishor