pal-robotics/aruco_ros #114: Process dies everytime I try to access the topic aruco_marker_publisher/result

jhielson, 2023-04-06:
Hi everyone,

I have been trying to recognize some aruco markers using your ros package but it is not working for me.

You can find my launch file here:
```
<launch>

    <arg name="markerSize"      default="0.05"/>    <!-- in m -->
    <arg name="ref_frame"       default="camera_link"/>  <!-- leave empty and the pose will be published wrt param parent_name -->

    <node pkg="aruco_ros" type="marker_publisher" name="aruco_marker_publisher">
        <remap from="/camera_info" to="/camera/rgb/camera_info" />
        <remap from="/image" to="/camera/rgb/image_mono" />
        <param name="image_is_rectified" value="False"/>
        <param name="marker_size"        value="$(arg markerSize)"/>
        <param name="reference_frame"    value="$(arg ref_frame)"/>   <!-- frame in which the marker pose will be refered -->
        <param name="camera_frame"       value="camera_rgb_frame"/>
    </node>

</launch>

```
My camera's topics:

```
/camera/rgb/camera_info
/camera/rgb/image_color
/camera/rgb/image_color/compressed
/camera/rgb/image_color/compressed/parameter_descriptions
/camera/rgb/image_color/compressed/parameter_updates
/camera/rgb/image_color/compressedDepth
/camera/rgb/image_color/compressedDepth/parameter_descriptions
/camera/rgb/image_color/compressedDepth/parameter_updates
/camera/rgb/image_color/theora
/camera/rgb/image_color/theora/parameter_descriptions
/camera/rgb/image_color/theora/parameter_updates
/camera/rgb/image_mono
/camera/rgb/image_mono/compressed
/camera/rgb/image_mono/compressed/parameter_descriptions
/camera/rgb/image_mono/compressed/parameter_updates
/camera/rgb/image_mono/compressedDepth
/camera/rgb/image_mono/compressedDepth/parameter_descriptions
/camera/rgb/image_mono/compressedDepth/parameter_updates
/camera/rgb/image_mono/theora
/camera/rgb/image_mono/theora/parameter_descriptions
/camera/rgb/image_mono/theora/parameter_updates
/camera/rgb/image_raw
/camera/rgb/image_raw/compressed
/camera/rgb/image_raw/compressed/parameter_descriptions
/camera/rgb/image_raw/compressed/parameter_updates
/camera/rgb/image_raw/compressedDepth
/camera/rgb/image_raw/compressedDepth/parameter_descriptions
/camera/rgb/image_raw/compressedDepth/parameter_updates
/camera/rgb/image_raw/theora
/camera/rgb/image_raw/theora/parameter_descriptions
/camera/rgb/image_raw/theora/parameter_updates
/camera/rgb/image_rect_color
/camera/rgb/image_rect_color/compressed
/camera/rgb/image_rect_color/compressed/parameter_descriptions
/camera/rgb/image_rect_color/compressed/parameter_updates
/camera/rgb/image_rect_color/compressedDepth
/camera/rgb/image_rect_color/compressedDepth/parameter_descriptions
/camera/rgb/image_rect_color/compressedDepth/parameter_updates
/camera/rgb/image_rect_color/theora
/camera/rgb/image_rect_color/theora/parameter_descriptions
/camera/rgb/image_rect_color/theora/parameter_updates
/camera/rgb/image_rect_mono
/camera/rgb/image_rect_mono/compressed
/camera/rgb/image_rect_mono/compressed/parameter_descriptions
/camera/rgb/image_rect_mono/compressed/parameter_updates
/camera/rgb/image_rect_mono/compressedDepth
/camera/rgb/image_rect_mono/compressedDepth/parameter_descriptions
/camera/rgb/image_rect_mono/compressedDepth/parameter_updates
/camera/rgb/image_rect_mono/theora
/camera/rgb/image_rect_mono/theora/parameter_descriptions
/camera/rgb/image_rect_mono/theora/parameter_updates
/camera/rgb_debayer/parameter_descriptions
/camera/rgb_debayer/parameter_updates
/camera/rgb_rectify_color/parameter_descriptions
/camera/rgb_rectify_color/parameter_updates
/camera/rgb_rectify_mono/parameter_descriptions
/camera/rgb_rectify_mono/parameter_updates

```
I can launch the package without error:

```
roslaunch aruco_ros marker_publisher.launch 
... logging to /home/trust/.ros/log/aca6f10a-d474-11ed-89e1-3bc6d567e23e/roslaunch-trust-Precision-7520-46642.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://trust-Precision-7520:44493/

SUMMARY
========

PARAMETERS
 * /aruco_marker_publisher/camera_frame: camera_rgb_frame
 * /aruco_marker_publisher/image_is_rectified: False
 * /aruco_marker_publisher/marker_size: 0.05
 * /aruco_marker_publisher/reference_frame: camera_link
 * /rosdistro: noetic
 * /rosversion: 1.15.15

NODES
  /
    aruco_marker_publisher (aruco_ros/marker_publisher)

ROS_MASTER_URI=http://localhost:11311

process[aruco_marker_publisher-1]: started with pid [46663]
[ WARN] [1680787470.967149869]: length of camera_info D vector is not 4, assuming zero distortion...

```

But, when I try to read the output topic (/aruco_marker_publisher/result
), aruco_ros dies:
```
[aruco_marker_publisher-1] process has died [pid 47347, exit code -11, cmd /home/trust/ros_ws/devel/lib/aruco_ros/marker_publisher /camera_info:=/camera/rgb/camera_info /image:=/camera/rgb/image_mono __name:=aruco_marker_publisher __log:=/home/trust/.ros/log/727f1fe2-d47f-11ed-89e1-3bc6d567e23e/aruco_marker_publisher-1.log].
log file: /home/trust/.ros/log/727f1fe2-d47f-11ed-89e1-3bc6d567e23e/aruco_marker_publisher-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

``` 

Can you please help me?

saikishor, 2023-04-06:
Hello @jhielson,

Thank you for writing to us. Could you please tell us on which version of ROS are you trying to test the code? Have you tested the same in a clean docker environment?

Thank you,

Best Regards,
PAL Robotics Team.

jhielson, 2023-04-06:
Hi @saikishor,

Thanks for your prompt reply. I am using ros noetic without docker. 

Best regards,
Jhielson M. Pimentel

jhielson, 2023-04-06:
Hi @saikishor,

Which versions do you recommend for Ubuntu, ROS and OpenCV?

Best regards,
Jhielson M. Pimentel

saikishor, 2023-04-06:
Hello @jhielson,

> Which versions do you recommend for Ubuntu, ROS, and OpenCV?
Standard Ubuntu with ROS noetic should be fine. On noetic, it works with version 4.2.0.

We are unable to reproduce the issue. Could you give us more information on this?

Thank you,

Best Regards,
PAL Robotics Team.

jhielson, 2023-04-06:
Hi,

I have been using Ubuntu 20.4, ROS noetic and OpenCV 4.7.0. 
Do you recommend any docker image?

I tried to use the aruco package with my web_cam (http://wiki.ros.org/usb_cam) and kinect (http://wiki.ros.org/freenect_launch). But it did not work for both of them. 

I can launch everything and see on rviz the camera's image.

[image]

But aruco_ros dies when I try to read the result topic:

[image]

I do not know how to solve this issue.

Best regards,
Jhielson M. Pimentel

closed 2023-04-06 by saikishor