pal-robotics/aruco_ros #117: [bug] double free detected in tcache 2

cyborgdennett, 2023-06-29:
Hi, I try to run aruco ros with the marker_publish launch file

I run it and this is my output

```
ros2 launch my_package aruco_launch.py
[INFO] [launch]: All log files can be found below /home/casper/.ros/log/2023-06-29-23-58-33-329583-LAPTOP-0CAT7HLP-575442
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [marker_publisher-1]: process started with pid [575446]
[marker_publisher-1] [INFO] [1688075913.954834708] [marker_publisher]: Waiting for the camera info...
[marker_publisher-1] [INFO] [1688075913.999797729] [marker_publisher]: Successfully obtained the camera info!
[marker_publisher-1] [INFO] [1688075914.032808398] [marker_publisher]: Successfully setup the marker publisher!
[marker_publisher-1] free(): double free detected in tcache 2
```

system:
- ROS2 Humble
- WSL Ubuntu 22.04
- Installed aruco_ros with rosdep

cyborgdennett, 2023-06-30:
https://github.com/pal-robotics/aruco_ros/blob/humble-devel/aruco_ros/launch/marker_publisher.launch.py

gives an error

But running

https://github.com/pal-robotics/aruco_ros/blob/humble-devel/aruco_ros/launch/single.launch.py

Does work for me.

saikishor, 2023-08-04:
Hello @cyborgdennett,

We will try to fix it at the earliest. If you have fixed it at your end, please feel free to open a PR. We will be happy to merge it.

Thank you,

Best Regards,
PAL Robotics Team.