pal-robotics/aruco_ros #118: [Humble] Failing to build on buildfarm because of ambiguity

audrow, 2023-08-08:
It seems that their is ambiguity about which overload is being called for the subscribe function. 

You can see this in the buildfarm here:
https://build.ros2.org/view/Hbin_uJ64/job/Hbin_uJ64__aruco_ros__ubuntu_jammy_amd64__binary/21/console

Maybe this has to do with the last patch release, which happened around the time this job started failing.

saikishor, 2023-08-08:
Hello @audrow,

Yes, we noticed it a few days back. We couldn't address it as most of the team is on vacation. I will try to fix it during these days. Thanks for the heads-up anyway.

Thank you,

Best Regards,
Sai

saikishor, 2023-08-08:
> It seems that their is ambiguity about which overload is being called for the subscribe function.

Hello @audrow!

I've taken a quick look, it seems like this PR has broken most of it's downstream packages and thereby causing ambiguity:
https://github.com/ros-perception/image_common/pull/276.

I'll see what I can do.

Thank you,

Best Regards,
PAL Robotics Team

saikishor, 2023-08-14:
Hello @audrow,

The package of image_transport is now released to upstream and it seems like it already fixed the CI: https://build.ros2.org/job/Hbin_uJ64__aruco_ros__ubuntu_jammy_amd64__binary/29/. Closing the issue as it is already fixed.

Thank you for your patience,

Best Regards,
PAL Robotics Team.

closed 2023-08-14 by saikishor