pal-robotics/aruco_ros #121: cannot show position,but get marks > Hello @Yaoxingtian,

Yaoxingtian, 2023-09-28:
> Hello @Yaoxingtian,
> 
> Thank you so much for your interest in the package. The estimation of ARuCo is based on the camera info topic, without having this info it is not possible to estimate the pose of the marker. That's why, if your camera is well calibrated, it will be reflected in the `camera_info`; hence, you have a better estimation.
> 
> We hope this answers your question.
> 
> Thank you,
> 
> Best Regards, PAL Robotics Team.

thanks so much for you kindly reply !
and now i add camera info as a topic ,when i run single.launch,it can gets marks but can not show the position on the picture,but echo /aruco_single/pose have enough information, i'am so confused , could you pls. help ?
single.launch:
[image]

/aruco_single/pose:
header: 
  seq: 46
  stamp: 
    secs: 1695872446
    nsecs: 169000149
  frame_id: "stereo_gazebo_left_camera_optical_frame"
pose: 
  position: 
    x: -30.77961287838618
    y: 0.9469890380327535
    z: 0.6522766351699829
  orientation: 
    x: 0.7413994447944224
    y: -0.6668339357097456
    z: -0.01968394403844307
    w: 0.07260790446958565
---

_Originally posted by @Yaoxingtian in https://github.com/pal-robotics/aruco_ros/issues/120#issuecomment-1738394662_

Yaoxingtian, 2023-09-28:
[image]

saikishor, 2023-09-28:
Hello @Yaoxingtian!

I'm glad that after adding the information, you could see the position after adding the camera_info. Regarding the marker, yes it should show it on top of the image, however, if you need position, you are supposed to subscribe to the topic and compute it. We haven't done it as most of the time it is redundant.

[image]

Thank you,

Best Regards,
Sai Kishor

closed 2023-10-03 by saikishor