pal-robotics/aruco_ros #125: the direction changes a lot when i put quaternion ioto navigation

Yaoxingtian, 2023-10-30:
hi, i meet a big problem that i cannot explain.
when i get quaternion for robot navigation, i find the direction always changes a lot ,not stable.no matter how big the code is.

do you know what's the proble ?  thanks so much!

https://github.com/pal-robotics/aruco_ros/assets/21310066/7059fe6d-aff8-4fcf-89cf-03efc5e1009a

saikishor, 2023-10-30:
Hello @Yaoxingtian!

Thanks for writing to us. Regarding your issue, I suggest using a rectified image in this case, and adding some filters to the pose would also help. You can also use 2 ArUco markers to distinguish the orientation. I hope this helps.

Thank you,

Best Regards,
PAL Robotics Team.

Yaoxingtian, 2023-10-31:
thanks for your advice !
and what do you think the valid Camera Resolution for stably detection？