pal-robotics/aruco_ros #130: why the size of distorsionCoeff is 4?

xibeisiber, 2024-04-03:
Hi, the distorsion coeff in sensor_msgs/CameraInfo has size of 5. 

But in [aruco_ros_utils.cpp](https://github.com/pal-robotics/aruco_ros/blob/humble-devel/aruco_ros/src/aruco_ros_utils.cpp), only 4 elements is used in distorsionCoeff.

why the last element is not used?

Thanks!

saikishor, 2024-04-20:
Hello @xibeisiber !

A very good question, I'm pretty sure there is a reason for it. I've to take a look and get back to you.

Thank you

saikishor, 2024-04-20:
Hello @xibeisiber!

I remember now, what we have observed is in general for normal camera models when the rectified image is parsed it usually works without any issues and the estimation is good. In the case of fisheye cameras, we have 4 distortion parameters and we usually undistort when we use a fisheye camera. If you want to add support for other cameras as well, please go ahead and open a PR, we will be happy to review and get it merged.

Thank you,

Best Regards,
PAL Robotics Team

xibeisiber, 2024-04-22:
> I remember now, what we have observed is in general for normal camera models when the rectified image is parsed it usually works without any issues and the estimation is good.

For a normal camera, if I use a raw image and set "image_is_rectified" to false, aruco will not actually accept the distortion coefficient, since all distortion elements are set to 0 if the coefficient size is not 4.

Is this a problem?
Thanks!

xibeisiber, 2024-04-28:
I assume that it depends on how much the image is distorted. 

If the distortion coefficient is relatively larger for a normal camera, wouldn't the estimation be improved more or less if the distortion coefficient is considered? 

Thanks!
xibeisiber