pal-robotics/aruco_ros #133: noetic process has died

Panda12580, 2024-04-20:
Hello, may I ask why when I run single,launch file, call rqt to see the result will kill the process? I have checked the set parameters and there is no error. I ask for your help.
[image]

saikishor, 2024-04-20:
Hello @Panda12580!

It is indeed weird that setting parameters through service works but not through the rqt reconfigure, at the end of the day, they are doing the same. We cannot see the screenshot you have attached. Could you reattach it?

Thank you

Panda12580, 2024-04-21:
@saikishor Thank you for your reply. I have updated and uploaded the picture. Please check where my mistake is.

Panda12580, 2024-04-21:
I set up the single.launch file and then start rs_camera.launch, but I want to see images via rqt and subscribed to aruco_ros/result , 
 the background process will be killed as soon as I do this.

saikishor, 2024-04-24:
Can you let us know what happens if you subscribe through command line?

Oscar-qi, 2024-04-24:
I opened three terminals and ran the driver node for the d435 camera, the single_realsense node for aruco_ros, and subscribed to aruco_single/result using the command line。

> The terminal display for subscribing to aruco_single/result using the command line reads:
rosrun image_view image_viewer  image:=/aruco_single/result
[INFO] 【1713951282.914114320]: Inttlaltzing nodelet wtth 4 worker threads.
[INFO][1713951283.689522413]: Using transport "raw"

> When I subscribe to the aruco_single/result topic, it shows the following in the terminal of aruco_ros:

NODES
aruco_single (aruco_ros/stngle)
ROS_MASTER_URI=http://Localhost:11311
process[aruco_stngle-1]:started with pid[80918]
[ WARN] [1713951115.318433478]:Corner refinenent opttons have been removed in Aruco 3.0.0, corner_refinenent ROS paraneter ts deprecated
[ WARN] [1713951116.014256637]: normaltzeImagerllumtnatton is untnplemented!
[aruco single-1] process has dled [pid 80918, extt code -l1, cmd /home/rm/catkin _ws/devel/lib/aruco_ros/single /camera_info:=/camera/color/canera_info   /image:=/camera/color/image_raw __name:=aruco_single __log:=/home/rm/.ros/log/451bc2ea-021d-11ef-8aea-48b02d15ce9b/aruco single-1.log].
log file：/home/rm/.ros/log/451bc2ea-021d-11ef-Baea-48bo2d15ce9b/aruco single-1*
.1og
all processes on machine have died, roslaunch will exit shuttting down processing monitor...
... shutting down processing monitor complete
done

Oscar-qi, 2024-04-24:
> Can you let us know what happens if you subscribe through command line?

I opened three terminals and ran the driver node for the d435 camera, the single_realsense node for aruco_ros, and subscribed to aruco_single/result using the command line。

The terminal display for subscribing to aruco_single/result using the command line reads:
rosrun image_view image_viewer image:=/aruco_single/result
[INFO] 【1713951282.914114320]: Inttlaltzing nodelet wtth 4 worker threads.
[INFO][1713951283.689522413]: Using transport "raw"

When I subscribe to the aruco_single/result topic, it shows the following in the terminal of aruco_ros:

NODES
aruco_single (aruco_ros/stngle)
ROS_MASTER_URI=http://Localhost:11311
process[aruco_stngle-1]:started with pid[80918]
[ WARN] [1713951115.318433478]:Corner refinenent opttons have been removed in Aruco 3.0.0, corner_refinenent ROS paraneter ts deprecated
[ WARN] [1713951116.014256637]: normaltzeImagerllumtnatton is untnplemented!
[aruco single-1] process has dled [pid 80918, extt code -l1, cmd /home/rm/catkin _ws/devel/lib/aruco_ros/single /camera_info:=/camera/color/canera_info /image:=/camera/color/image_raw __name:=aruco_single __log:=/home/rm/.ros/log/451bc2ea-021d-11ef-8aea-48b02d15ce9b/aruco single-1.log].
log file：/home/rm/.ros/log/451bc2ea-021d-11ef-Baea-48bo2d15ce9b/aruco single-1*
.1og
all processes on machine have died, roslaunch will exit shuttting down processing monitor...
... shutting down processing monitor complete
done

chenglei11, 2024-05-27:
hi I ran into the same problem with you, have you solved it? could you please tell me how to solve this one? you can reply me in Chinese as well

Panda12580, 2024-05-27:
@chenglei11 I was able to implement it on another noetic version of the computer, but the code used was previous, that is, I'm not sure whether the author of the package changed the code to cause the conflict or the computer environment itself, so I guess if I can use the earlier version of the noetic package to try it?

chenglei11, 2024-05-29:
> @chenglei11 I was able to implement it on another noetic version of the computer, but the code used was previous, that is, I'm not sure whether the author of the package changed the code to cause [...]

yeah i can also implement it on another computer, have you tried this method? did it work?

BUAAPY, 2024-10-23:
> @chenglei11 I was able to implement it on another noetic version of the computer, but the code used was previous, that is, I'm not sure whether the author of the package changed the code to cause [...]

Hey, I've met the same problem. Have you solved the problem? which version have you tried?

Panda12580, 2024-10-23:
I'm using the noetic branch, and the code I'm using is from earlier, on my other computer. Unfortunately, my rough review of the new code differs greatly from the current one. So I'm not sure what the problem is, the environment configuration problem is very likely.

Panda12580, 2024-10-23:
@BUAAPY I'm using the noetic branch, and the code I'm using is from earlier, on my other computer. Unfortunately, my rough review of the new code is not very different from the current one. So I'm not sure what the problem is, the environment configuration problem is very likely.

LuShuaiqi-Geo, 2024-11-12:
I encountered the same problem. My image acquisition operations were the same as the author's. I also used neotic. But the difference is that after I tried to get the image, it did not kill aruco_single, but got stuck. It could not get the image and did not kill the process. I feel very strange and hope there is a solution to this problem

BUAAPY, 2024-11-12:
I've tried to print messages during the running process and the program has stucked before an opencv function (the resize function). However, the program runs smoothly in another noetic environment. The only difference is the version of opencv (4.5.4  in the stucked one, 4.2.0 the other one). 

According to other blogs the ROS noetic uses its default opencv (4.2.0), especially its cv_bridge package. The aruco_ros uses my installed opencv (4.5.4) automatically during the cmake process, though the minimum version is set to 4.2.0 in the cmake file. I dont know if this is the issue and how to fix it. Maybe uninstalling the latest opencv and using the default one o noetic may help.

Panda12580, 2025-01-06:
@saikishor It doesn't seem right either. I set the opencv version to 4.2 during the compilation process but the problem still occurs. I default to 4.6, then I install 4.2 and change the desired opencv version to 4.2 in the cmakelists file in aruco_ros, and I print out to confirm that I'm using 4.2 in singe.cpp but still get errors. So we sincerely hope that you can help us solve this problem！😊

andrewhutong12345, 2025-01-10:
@Panda12580 The `aruco_ros` package actually uses `cv_bridge` to find opencv, so you should make sure the cv_bridge is point to the right opencv.
For me, the problem is that when I install the ros-noetic, the version of the default opencv is 4.5.2, not 4.2.0 . I guess maybe the reason is that my platform is JETSON ORIN. So just make sure the opencv' s version is 4.2.0 and cv_bridge is indeed point to that opencv!

Panda12580, 2025-01-10:
@andrewhutong12345 Yes, I think you make a lot of sense, I just print in the single.cpp file whether the program is running with cv4.2 ignoring the cv_bridge pointing problem, I'll try to see if I can fix it.

saikishor, 2025-01-10:
Hello! 

Sorry for not being able to respond earlier. I'll see if I can find time to be able to add something to the code, such that if the opencv version is not the expected one. It fails more explicitly with a proper message. 

It would be nice to see the difference between 4.2.0 and 4.5.x and see if we can have it working in both. If someone is interested to take a look, feel free to open a PR. I'll be happy to review and get it merged.

Thank you

307321587, 2025-01-15:
> @Panda12580 The `aruco_ros` package actually uses `cv_bridge` to find opencv, so you should make sure the cv_bridge is point to the right opencv. For me, the problem is that when I install the [...]

This reason is correct. I find that the cv_bridge .cmake file in the path /opt/ros/noetic will set libraries paths such as "libopencv4.2xx". If the local opencv is not 4.2 or the opencv library is "libopencv_world", the program will died without the libraries. 
The solution is simple. I removed the cv_bridge and download its source code in github. https://github.com/ros-perception/vision_opencv. Remember change the branch to noetic. Then put the source code in the workspace with aruco lib and compile them togther, the problem is solved.

`
sudo apt-get remove ros-melodic-cv-bridge
git clone https://github.com/ros-perception/vision_opencv.git
cd vision_opencv
gitcheckout noetic
catkin_make
`

These reference wedsites also guided me: 
https://zhuanlan.zhihu.com/p/392939687
https://www.cnblogs.com/gooutlook/p/18394497