pal-robotics/aruco_ros #134: pose representation problem

waliwali1111, 2024-04-26:
Hello, is there a function in the function package that allows the user to choose the representation method of pose? For example, should quaternion numbers or Euler angles be used? How to operate?

saikishor, 2024-05-09:
Hello @waliwali1111!

Right now, there is no such functionality. Feel free to add utility methods in the PR

Thank you,

Best Regards,
PAL Robotics Team

Tanishq30052002, 2025-04-08:
Hi @saikishor,
since there is no activity on the current issue, can i continue with it?

Tanishq30052002, 2025-04-21:
@saikishor can you explain what the expected output from this is?

saikishor, 2025-04-21:
> @saikishor can you explain what the expected output from this is?

I believe @waliwali1111 is talking about the aruco library itself. Thinking about it again, i think it's better not to touch that part as it is a part of the library and adding things is just going to be harder in future. Unless we have a real good usecase, I wouldn't go ahead with any further chances. 

Thank you @Tanishq30052002 for pinging again

closed 2025-04-21 by saikishor