pal-robotics/aruco_ros #139: makers not detected in gazebo camera images

david-wb, 2025-01-09:
I'm trying to detect makers in images from a simulated camera in gazebo ignition (fortress). Aruco is receiving the images and camera info, however the markers are not detected. The exact same setup works when using a real camera (astra), but not in simulation. Any ideas what the problem might be? I have a hunch it is related to either lighting conditions or the camera_info data. Thanks in advance.

[image]

The simulated camera feedback you see in the bottom left of the screenshot above shows the "aruco result" image which has no markers.

david-wb, 2025-01-09:
I should add that I'm using the `marker_publisher` launch file. I have verified the image callback is called but not makers are detected. Thus, I think it must be a problem with either the image or the camera params.

Ayush-M99, 2025-03-06:
> I'm trying to detect makers in images from a simulated camera in gazebo ignition (fortress). Aruco is receiving the images and camera info, however the markers are not detected. The exact same setup [...]

Hi,
I have had issues of aruco markers being undetected in the past. Several techniques/reasons were:-
Solutions:-
1. Create a grayscale mask on the input images for bettet clarity of aruco markers in most lighting conditions.
2. Sometimes the camera might not detect cause the input size of the markers might be quite small, I implemented a zoom attribute to the input image which has scales my image to 2-2.5 times the original.

Hope this helps!! :)