pal-robotics/aruco_ros #141: Cannot launch any of 3 node types on ROS Noetic and Ubuntu 20.04

ScheglyakStas, 2025-03-25:
I'm using ROS Noetic on Ubuntu 20.04.
I can't launch any of the 3 node types (single, double, marker_finder) in the package using roslaunch.
The error displayed is:
`ERROR: cannot launch node of type [aruco_ros/single]: Cannot locate node of type [single] in package [aruco_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)`
**However,  I have already set the C++ files as executable with `chmod +x`, built the package and sourced `devel/setup.bash`.**
`catkin_make clean` does not help, neither does reinstallation of the package from `noetic_devel` branch.

rajRishi22, 2025-03-26:
Hi @ScheglyakStas  , can you help me to reproduce this issue , i want to contribute to it?

saikishor, 2025-03-26:
@ScheglyakStas Did you try sourcing your local workspace after compiling?

ScheglyakStas, 2025-03-29:
Thanks everyone, I've solved it. I deleted the build and devel folders manually, and tried to build the project again. However, an unused package produced a build error. I deleted that package, rebuilt the project, and launched the Aruco nodes without any issues.

closed 2025-03-29 by ScheglyakStas